tzapu/WiFiManager #1: False EEPROM reading after manual restart

renno-bih, 2015-06-11:
This library is exactly what I need but looks like it doesn't work as it should.

Module starts in AP mode. After entering SSID and password it automatically restarts and reads correct data from EEPROM and successfully connects to my WiFi router.

But as soon as I manually restart ESP-01 it reads false information from EEPROM, refuse to connect and starts AP again.

Here is what I get when restart my ESP manually:

```
AutoConnect
Reading EEPROM SSID
SSID: 
��?�   ntX     <script>function
Reading EEPROM Password
Password: 
 c(l){document.getElementById('s').value=l.innerText||l.textCont
Waiting for Wifi to connect
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.

Could not connect to WiFi
Started Soft Access Point

WiFi connected
0.0.0.0
192.168.4.1
mDNS responder started
Server started
```

tzapu, 2015-06-15:
hi, didn t actually test it on and esp01, just esp12 (nodemcu).
it kind of looks to me like some kind of memory issue, like the costants are overwriting the part of memory where the eeprom is. i ll take a look at that.

in the mean time, maybe rewrite the cost strings in your library to the minimum needed? remove all css, etc from them, just the absolute bare minimum.

tzapu, 2015-06-15:
might be related to https://github.com/esp8266/Arduino/issues/414 
description seems similar

lesjaw, 2015-07-04:
any fix on this? tzapu can you make a libary for this? https://github.com/jeffryr/ESP8266-Sketches/tree/master/Onboard 

it works on esp01... esp01 seem have trouble saving to eeprom..

thanks

tzapu, 2015-07-06:
i don t know if there s any fix from the arduino guys, but i have tested on an esp 01 and it works without any issues with the latest staging release of the esp arduino board manager package

i want to extend this further and using the auto saved values wouldn t work well for me, but i will ponder it further

cheers

lesjaw, 2015-07-06:
ah ok..will test this again with lates stag.. btw join our mailing list on souliss.. https://github.com/souliss/souliss/wiki

lesjaw, 2015-07-08:
i have tested with the latest ide of arduino, it seem still dont work, it work on reset, but it didnt work after unplug the power supply..

tzapu, 2015-07-09:
this is so weird @lesjaw , did you test on a couple of esp s or on just one?
i have personally tested and deployed this on 2 nodemcu 0.9, 1 nodemcu 1.0 and 1 esp-01 and seems fine.

could you send me you sketch so i can compile that and see what happens? is it a very large sketch or allocating memory/eeprom in some other way?

i like souliss, looked at it a few times, but since i m an ios user can t do much with it.

lesjaw, 2015-07-09:
oh you use nodemcu firmware, i use v0.9.5.2 AT Firmware, souliss have openhab binding, so you can use it in iOS, I use openhab for iOS user interface..

this is the sketch http://pastebin.com/dLM7Bbnq

as you can see, i use dynamic addressing in souliss, it will detect the address and save it to eeprom too.. and it didn't work too on esp after power unplug.. i tried on esp01 and 12 with 0.9.5.2 firmware...

this is souiss library for store.h https://github.com/souliss/souliss/blob/friariello/tools/store/store.h#L14

since your wifi manager lenght save 64 byte, i change the store index of souliss to 65..

this is the store.cpp https://github.com/souliss/souliss/blob/friariello/tools/store/store.cpp

lesjaw, 2015-07-12:
Omg it's me who stupid... Sorry.. 

It's been month since I use esp arduino board... So today I update it... Now its working flawless... Perfect!.. 

Btw it's confuse me, you are using nodemcu firmware with arduino ide?

tzapu, 2015-07-13:
hi, glad it s not the library :p
i am using just the nodemcu board, arduino ide built firmware. i fine nodemcu 0.9 board very easy to use with arduino, you don t have to use the nodemcu firmware on it
https://github.com/nodemcu/nodemcu-devkit

glad it s now working
cheers

closed 2015-07-13 by tzapu

lesjaw, 2015-07-13:
oh i see, btw your library now become default in souliss.. nice..

tzapu, 2015-07-13:
why thank you very much, i am honoured

juanpintom, 2015-07-13:
I want to say Thank you too, it was really easy to integrate this useful library on Souliss :)

Regards!