tzapu/WiFiManager #2: You have an error in WifiManager.cpp

myk3y, 2015-06-18:
Your library is called WiFiManager, your files are called WifiManager.h and WifiManager.cpp and the references in WifiManager.cpp - L12 - are for #include "WiFiManager.cpp"

I can't understand how this tested correctly?

myk3y, 2015-06-18:
I corrected the error and ran the code as per your instructions - and it doesn't work at all.

Instead of an AP called 'ESP-AP', you end up with one called NoNetESP and nothing sensible on serial.

tzapu, 2015-06-18:
hello, my bad, updated the lib but not the readme.
the file names weren't a problem, but i've updated those as well in the latest commit for consistency. thanks.

if you can, fetch the latest one and do the following 
include 
#include <WiFiManager.h>
init
WiFiManager wifi(0);
setup function
wifi.autoConnect("ESP-AP");

that should start the access point mode if it doesn t connect to wifi. you can then connect to it and open 192.168.4.1 in a browser

myk3y, 2015-06-18:
Cool, thanks.

closed 2015-06-18 by myk3y