tzapu/WiFiManager #3: Should redirect all IP addresses to configuration page

probonopd, 2015-07-07:
WiFiManager should redirect all IP addresses to configuration page so that the user does not need to specifically know 192.168.4.1.

See https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino

tzapu, 2015-07-09:
good idea, thanks for the link, will look into adding this

tzapu, 2015-09-11:
added captive portal

closed 2015-09-11 by tzapu

probonopd, 2015-09-12:
Thank you. I will test this as soon as it compiles with the official ESP8266 package installed by the Boards Manager. Right now compilation fails with `error: 'class ESP8266WebServer' has no member named 'hostHeader'` which is expected since `hostHeader`was added only recently to  esp8266/Arduino.