tzapu/WiFiManager #4: allow custom parameters to be configured

tzapu, 2015-08-01:
so you can use the interface for other things than wifi user and password

Sylvania2, 2015-08-01:
Would be nice to save BLYNK token this way

tzapu, 2015-10-07:
or emoncms tokens :)
i am thinking on moving to FS support instead of EEPROM before implementing this.

nibelungen, 2015-11-02:
+ on that.
Other ideas to save:

pre defined SSIDs and PWs make use of known networks such as hotspots.
Way to implement - retry different/saved credentials before going to AP mode.

Maybe up to 5 pre defined Credentials.

AndersV209, 2015-11-18:
@tzapu any idea on when you will add FS support ?

tzapu, 2015-11-19:
@AndersV209 based on the end of discussion here https://github.com/esp8266/Arduino/issues/849#issuecomment-146499499
I am going through with simplifying it first of all so it's easy for people just getting started to use it, in the spirit of Arduino.
I do have some thoughts to still offer advanced support like FS, but I'd like to sort the stuff on igrr's list first of all

tablatronix, 2015-12-16:
2 simple callouts/filters should suffice for this.

Jorgen-VikingGod, 2015-12-22:
Currently I add this "feature" as fixed code in my fork
because I need to configure MQTT host,port,user,pass

I try to add this as generic part - but currently no time left for this :/
We will a method to add/handle additional HTML code for the submit form
Or just simple method to add one entry by following parameters:
- name
- default value
- max length
- type

tzapu, 2015-12-22:
what i m thinking is to give it an array of params you need, and get a callback with them filled in

jeremyabel, 2015-12-23:
Would also be cool to be able to pass in our own css style string.

tzapu, 2015-12-25:
@jeremyabel will take it into consideration
thanks

rpseng, 2016-01-06:
Just sent a pull request #47 with a working version with custom parameters to be configured by the user.

Each parameter has an id, placehoder, default value and maximum length. It worked fine for me, check the example file.

All the best.

tzapu, 2016-01-07:
hi guys, i have modified a bit the pull request kindly provided by @rpseng  (THANK YOU!!!) so anyone can use their own storage mechanism for those parameters
I have added a FS storage example. Will add an EEPROM example soonish as well.
It is not as pretty as the original version nor as simple, but I really did not want to readd eeprom or FS as a dependency to WiFiManager
If anyone could take a look and let me know if they feel is ok or not, the branch is here
https://github.com/tzapu/WiFiManager/tree/dev

if not, i will merge it in a couple of days

rpseng, 2016-01-07:
Hi tzapu, I think your dev branch is fine! Now there is room for some cleanup, there are several variables/methods not used anymore.

Regarding the example using EEPROM, I believe it is very welcome. Very simple ESP-01 setups will benefit from it.

tzapu, 2016-01-07:
thanks @rpseng 
will cleanup, release this one, then use your code to make an eeprom version as well.
Thanks.

tzapu, 2016-01-08:
added custom parameters to main branch together with FS storage example
now need to fix conflicts with other pull requests :P

tzapu, 2016-01-13:
this has been added, examples are there, documentation is coming shortly

closed 2016-01-13 by tzapu

iranweld, 2017-11-25:
i want use wifimanger for this file
[link](https://gist.github.com/4refr0nt/829052635894da0f5715)
butt i cannat do it
i want read this parameter by wifimanger.h
String mqttServerName = "m11.cloudmqtt.com";            
int    mqttport = 10927;                               
String mqttuser =  "test";                            
String mqttpass = "test"; 
if possible plz help me