tzapu/WiFiManager #6: Sporadic resets

changannie, 2015-08-17:
When I attempt to connect to the soft access point, the module decides that it wants to reset to boot mode and then attempt autoconnecting again. This happens even when I'm running the example code.

Here's what the serial monitor displays when this occurs:

 ets Jan  8 2013,rst cause:4, boot mode:(3,5)

wdt reset
load 0x40100000, len 29156, room 16 
tail 4
chksum 0xea
load 0x3ffe8000, len 2604, room 4 
tail 8
chksum 0x03
load 0x3ffe8a30, len 4144, room 0 
tail 0
chksum 0x57
csum 0x57
r‚

Any idea why this happens or how I could fix this?

tzapu, 2015-08-26:
this happened to me on an esp01 after writing the firmware from the arduino ide. after a power cycle it seemed stable.

another possibility could be the wifi scan taking too long to complete. have you got a lot fo visible access points around you?

roccomuso, 2015-12-08:
Might be the peak consumption? Have you tried soldering a capacitor on the esp-01 VIN-GND?

teomaragakis, 2015-12-31:
This is usually a power issue, not a software one.

closed 2016-01-13 by tzapu

alwynallan, 2016-01-13:
Does "closed" mean no more comments?

I have a Sparkfun ESP8266 Thing that resets sometimes. I tracked it down to
Windows Update reboots, that cause the DTR & RTS lines to be jiggled in a
certain way that resets the ESP8266, so that it can be put in programming
mode without a button. The circuit is copied from NodeMCU, so those boards
probably have the same problem.

A. Peter Allan · 450 Timber Lane · Devon PA 19333 · 610-687-1708 ·
610-547-4836 mobile

tzapu, 2016-01-13:
hi, feel free to reopen or comment,
but it looks like most issues are either hardware or not specific to WiFiManager, so i can t really fix them :)