tzapu/WiFiManager #8: error: 'class ESP8266WebServer' has no member named 'hostHeader'

mattogodoy, 2015-10-25:
Hi!
Excellent library, thanks for sharing!

Although, I'm not able to compile it. I'm in a Mac using the original Arduino IDE and installed the ESP8266 libraries following the steps at: https://github.com/esp8266/Arduino

Then, I've added this library, but when I try to compile the example called «AutoConnect» I get this error:

```
Arduino: 1.6.5 (Mac OS X), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS)"

/Users/matto/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp: In member function 'boolean WiFiManager::captivePortal()':
/Users/matto/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:384:20: error: 'class ESP8266WebServer' has no member named 'hostHeader'
   if (!isIp(server.hostHeader()) ) {
                    ^
/Users/matto/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:386:82: error: 'class WiFiClient' has no member named 'localIP'
     server.sendHeader("Location", String("http://") + toStringIp(server.client().localIP()), true);
                                                                                  ^
Error compiling.
```

I've seen a guy with the same issue at:
http://www.esp8266.com/viewtopic.php?p=29587&sid=0ab2ce98b8ece51d93ae1806a5aaa92c#p29587

Any advice?
Thanks again and keep up the good work :)

tzapu, 2015-10-26:
HI, thanks for the kind words.

Did you try with the staging version of the library?

Cheers

mattogodoy, 2015-10-26:
Awesome!
That did the trick. Thanks a lot.

tzapu, 2015-10-27:
no problem. glad you can use it

closed 2015-10-27 by tzapu

mkeyno, 2015-11-24:
hi  
what is the staging version?
I have same problem and extract your library to the arduino library
 error: 'class ESP8266WebServer' has no member named 'hostHeader'
   if (!isIp(server.hostHeader()) ) {

'class WiFiClient' has no member named 'localIP'
     server.sendHeader("Location", String("http://") + toStringIp(server.client().localIP()), true);

what might be conflicted with currently installed library?

should I remove extra library ?

thanks for help

tzapu, 2015-11-25:
hi,
you just need to install the staging package of esp8266
info here 
https://github.com/esp8266/Arduino#staging-version-

mkeyno, 2015-11-25:
hi 
thanks for quick reply but I'm windows user and install  Arduino esp library by this link inserted to board manager 
http://arduino.esp8266.com/stable/package_esp8266com_index.json

is there any way so I could find what library  should be changed ?

BR

tzapu, 2015-11-26:
hi, i think you need to do this then:
- go to boards manager
- find esp8266 and remove
- go to preferences and replace `http://arduino.esp8266.com/stable/package_esp8266com_index.json` with `http://arduino.esp8266.com/staging/package_esp8266com_index.json`
- go again to boards manager 
- find and install 8266 

i hope this works for you

mkeyno, 2015-11-26:
hi buddy
I change it and worst things happens , not only any update although my Arduino wont start again until restore line in the preferences.txt file
any idea buddy

tzapu, 2015-11-26:
hi, sorry to hear that
i don t know what could of happened
you might want to report this on https://github.com/esp8266/Arduino 
maybe it s some issue with the windows build

Juppit, 2015-11-26:
@mattogodoy did you edit preferences from the arduino IDE, I mean:
File ->Preferences ?
Do not edit the file preferences.txt if it's not necessary.

SCKStef, 2015-11-28:
i had the same Problem, installing the staging version did the job :-)

Lawlet, 2015-12-04:
Maaaaan you are great!! thank you!! and also thank you for bring to us this incredible library!!!

tzapu, 2015-12-04:
:+1:  :) thanks

danielitp, 2016-05-13:
'class WiFiManager' has no member named 'getConfigPortalSSID'

danielitp, 2016-05-13:
error: 'class WiFiManager' has no member named 'getConfigPortalSSID'
Serial.println(myWiFiManager->getConfigPortalSSID());

error: 'class WiFiManager' has no member named 'setMinimumSignalQuality'

    wifiManager.setMinimumSignalQuality(50);

error: invalid conversion from 'void (*)(WiFiManager*)' to 'void (*)()' [-fpermissive]

   wifiManager.setAPCallback(configModeCallback);

C:\Users\itp\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:61:13: error:   initializing argument 1 of 'void WiFiManager::setAPCallback(void (*)())' [-fpermissive]

     void    setAPCallback( void (*func)(void) );

hofmeisterchrigu, 2017-04-19:
Hi there
I did what **tzapu** commented on 26 Nov 2015 and since then the Arduino IDE doesn't know the nodemcuv2 board anymore. I can't compile and upload any code since I made the changes.

Error:
`Board nodemcuv2 (plattform esp8266, paket esp8266) unknown.`

Any idea?

Thanks Chris