tzapu/WiFiManager #13: WifiMulti instead of single Wifi

nibelungen, 2015-11-19:
Thanks for the implementation of my other suggestions.
Here is a new one - maybe you can use WifiMulti.h instead of the single one.

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiMulti/WiFiMulti.ino

Best regards

tzapu, 2015-11-19:
Hi,
You could simply achieve that by using WifiMulti as in the setup part of the example, and having WiFiManager after it, if it's not connected. You would then have the preset multiple AP's and fallback if that didn't work. 
Since it's and already made and distributed with the core library, it wouldn t make much sense to duplicate it's functionality.
Cheers

nibelungen, 2015-11-20:
Thanks - really good support and feedback from you so far!

To duplicate the code was not my intention, but to replace it in your examples and use it as default way to connect as it does not care if there is one or multiple predefined connections.

tzapu, 2015-11-20:
Unfortunately there are some pretty big differences on how the two methods work
The default way of connecting is included by default in all sketches, WiFiMulti would need an additional library to be included. WiFiMulti is actually using the default wifi connect in the background anyway.
The normal way of connecting tries to connect and if it can't it releases control back to you (so i can switch to AP mode and start a webserver, etc), WIFiMulti is made to scan continuously for one of the known networks until it manages to connect.

tzapu, 2015-12-03:
i will close this for now, will ponder at it more in the future

closed 2015-12-03 by tzapu