tzapu/WiFiManager #14: Issue compiling

mlongo95, 2015-11-23:
I get this issue every time I try to compile with WiFiManager in my sketch

{
C:\Users\Longo\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:113:10: note: in expansion of macro 'DEBUG_PRINT'
     void DEBUG_PRINT(Generic text);
          ^
}

The code compiles once i remove everything WiFiManager related from the sketch, any help would be appreciated.

Thanks, M

tzapu, 2015-11-23:
hi,
what arduino ide are you using?

mlongo95, 2015-11-23:
Using IDE 1.6.5 and the newest staging ESP8266 Core Package

tzapu, 2015-11-24:
well, i m at a loss here... never ran into this
although there s a slim to no chance for this to be related, could you rename 'WiFiManager-master' to 'WiFiManager' ?

closed 2015-12-28 by tzapu