tzapu/WiFiManager #15: Using PubSubClient with WiFiManager

lunanigra, 2015-12-01:
Hello, can you give some hints how to use PubSubClient with your WiFiManager?

To init the PubSubClient a WiFiClient is given as parameter...
WiFiClient espClient;
PubSubClient client(espClient);

Works simple and fine. But how to retrieve a PubSubClient instance when using WiFiManager? If you recommend a different MQTT library it would be fine for me as well.

Thanks, Jan Christian

tzapu, 2015-12-03:
Hi Jan,
Just use it as you mentioned above.
```
WiFiClient espClient;
PubSubClient client(espClient);
```
WiFiManager is active only during the startup phase. after it connects, you can build your app as normal. 
WiFiClient does not have any direct connection or dependency on WiFiManager

tzapu, 2015-12-09:
@lunanigra is it ok? did you get it running in the end?

lunanigra, 2015-12-09:
Hi! Thanks for your support.
Don't know what I did wrong last time; but it works :-)

Not sure if already fixed... When I tested the WifiManager few weeks ago some Wifi networks got listed twice. Nothing serious...

tzapu, 2015-12-10:
hi, no problem,

have not seen that happening. if it becomes an issue please open another issue/

cheers

closed 2015-12-10 by tzapu

Jimbwlah, 2016-04-19:
Hi,

Thank you for your excellent WifiManager. I am having an issue with the pubsubclient where by it doesnt maintain it's connected state -

I have my code as so

`// MQTT PubSubClient instantiation
WiFiClient wclient;
PubSubClient client(wclient, "my.mqtt.server", 1883);

/*--------------------
    Setup the WiFi
---------------------*/
void setupWiFi() {
  // WiFimanager instance
  WiFiManager wifiManager;
  //sets timeout until configuration portal gets turned off
  wifiManager.setTimeout(300);
  // Setup the AP
  wifiManager.autoConnect("AHUB");
  delay(3000);
}

/*-------------------------
     Setup the hardware
--------------------------*/
void setup() {

  // Start the serial
  Serial.begin(9600);

  // Setup the wifi server
  setupWiFi();
}

/*-------------------
    The Main Loop
--------------------*/
void loop() {

  // Connected to WiFi
  if (WiFi.status() == WL_CONNECTED) {

    // We aren't connected to MQTT broker
    if (!client.connected()) {

      Serial.println("not connected to MQTT broker, trying connection"); // debug only

      // Connect to MQTT
      if (client.connect(MQTT::Connect("myhub")
                         .set_clean_session()
                         .set_will("status", "disconnected")
                         )) {

        // Report this
        Serial.println("hub connected");

        // Callback listener for agwa/feeds
        client.set_callback(callback);
        client.subscribe("commands");
      }
    }

    // We are connected to MQTT Broker
    if (client.connected()) {
      Serial.println("connected, looping"); // debug only
      // Continue PubSubClient loop
      client.loop();
    }
  }
}
`

... unfotunately, it keeps connected and client.connected() is reset when client.loop() is called.

Any ides?

Thanks

MyIoT01, 2016-05-16:
Hi Tzapu,

First of all, thank you for your library that gave all of us. I also can't work out pubsubclient with wifi manager. Can you give me an example how to work out with it. Below code that i gave executed but doesn't work. Any help will be greatly appreciated.

` #include`` <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
  #include <PubSubClient.h>

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

const char* mqtt_server = "myadress.com";

WiFiClient espClient;
PubSubClient espClient;

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
    
}

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if an 1 was received as first character
  if ((char)payload[0] == '1') {
    digitalWrite(BUILTIN_LED, LOW);   // Turn the LED on (Note that LOW is the voltage level
    // but actually the LED is on; this is because
    // it is acive low on the ESP-01)
  } else {
    digitalWrite(BUILTIN_LED, HIGH);  // Turn the LED off by making the voltage HIGH
  }

}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP8266Client")) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("outTopic", "hello world");
      // ... and resubscribe
      client.subscribe("inTopic");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);
    client.setServer(mqtt_server, 1883);
    client.setCallback(callback);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
    if (!client.connected()) {
    reconnect();
  }
  client.loop();

  long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
    snprintf (msg, 75, "hello world #%ld", value);
    Serial.print("Publish message: ");
    Serial.println(msg);
    client.publish("outTopic", msg);
  }

}

``

fieldbob, 2016-06-15:
Ho tzapu. 
Thanks for your wifimanager.
I had the problem. The mqtt connection is not reliable with WifiManager.
My code is almost same abov except  i added client.loop() at the end.

Bob

LeonVos, 2017-07-15:
This should work: (ref to your MQTTserver)

WiFiClient wclient;
PubSubClient client(wclient, mqttserver);

amrithmmh, 2019-05-31:
> Hi! Thanks for your support.
> Don't know what I did wrong last time; but it works :-)
> 
> Not sure if already fixed... When I tested the WifiManager few weeks ago some Wifi networks got listed twice. Nothing serious...

can you share your code ?...how to get it working?

amrithmmh, 2019-05-31:
> This should work: (ref to your MQTTserver)
> 
> WiFiClient wclient;
> PubSubClient client(wclient, mqttserver);

does not work .It throws error

mediaDS, 2021-01-25:
Hi

Neither

`PubSubClient client(espClient);
`
nor

`PubSubClient client(wclient, mqttserver);
`
works. Both throw lenghty error messages. I am using PlatformIO and a ESP8266 board.

If I use

`PubSubClient client;`

it compiles without erros but the code crashes as soon as I enter the reconnect loop. The ESP8266 just restarts. So, is there any way to use both libs together?