tzapu/WiFiManager #16: eeprom

domonetic, 2015-12-03:
share settings with other data

from that address writes data occupy and how to use EEPROM to store other data?

tzapu, 2015-12-03:
hi, this was intially included, and there was an eeprom start paramter passed on, but i believe it is no longer functional.
I am planning to drop eeprom requirement completely, hopefully before the end of this week. you would then be able to use eeprom as needed

domonetic, 2015-12-03:
why not use spiffs to store config file? i think, this is more userful.

any ways is good job and simplifi initial configs

tzapu, 2015-12-04:
well, i'm trying to simplify it as much as possible, and giving options like these for advanced uses.
it's just a matter of getting the time and learning enough to make it a reality :p

domonetic, 2015-12-04:
I appreciate your work, and is useful. I would like to work with you.

Aprecio tu trabajo y es de real utilidad. Me gustaria colaborar contigo.

tzapu, 2015-12-05:
hi, that s nice to hear. please feel free to discuss and/or submit as pull request any improvements you might think fit

tzapu, 2015-12-05:
i ve just commited a new version that does not need eeprom or wificlient anymore
check it out

domonetic, 2015-12-05:
Great. Very efficient . Do you use the  section of config of wifi. Thanks.

closed 2015-12-12 by tzapu