tzapu/WiFiManager #17: error: field '_server' has incomplete type

roccomuso, 2015-12-08:
My error during compiling process:

    Arduino:1.6.6 (Windows 10), Scheda:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

    In file included from C:\Users\Rocco\Documents\Arduino\hello_world\hello_world.ino:4:0:

     C:\Users\Rocco\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:143:15: error: field '_server' has incomplete type

       WiFiServer  _server;

                   ^

    exit status 1

tzapu, 2015-12-08:
hi, are you using the latest staging release of the arduino esp8266 core?

roccomuso, 2015-12-08:
@tzapu I'm using the last 2.0 stable version:      http://arduino.esp8266.com/stable/package_esp8266com_index.json

roccomuso, 2015-12-08:
@tzapu Just discovered that if i move down the ESP8266WebServer.h dependency, in this way:

    #include <EEPROM.h>
    #include <WiFiClient.h>
    #include <DNSServer.h>
    #include <WiFiManager.h>
    #include <ESP8266WebServer.h>

I have no compile error.

tzapu, 2015-12-08:
cool,
thanks for letting me know.

you can also remove 
```
#include <EEPROM.h>
#include <WiFiClient.h>
```
if you got the latest version of the lib. examples still need updating

roccomuso, 2015-12-09:
@tzapu ok thank you.

tzapu, 2015-12-09:
@roccomuso does it work ok with the 2.0.0 stable version?

roccomuso, 2015-12-11:
@tzapu Tried today. It seems to work pretty well. :) I've tried it on a ESP866-01 board. And i've noticed that after the upload through the arduino ide (and GPIO0 pull down), you would get a <code>ets Jan  8 2013,rst cause:1, boot mode:(1,7)</code> error. So you must disconnect the pull down on GPIO0 and reset the board if you wanna make it works..

tzapu, 2015-12-11:
hi, that s pretty much general for all esp8266 modules. otherwise they reboot back into firmware upload mode.

glad it s working, i ll close this then

closed 2015-12-11 by tzapu