tzapu/WiFiManager #18: WifiManager PubSubClient and DHT22 sensor

roccomuso, 2015-12-08:
Hi,

Here is the code I'm trying to execute:

    #include <DNSServer.h>
    #include <WiFiManager.h>
    #include <ESP8266WebServer.h>

    #include "./libraries/DHT.h"
    #include <ESP8266WiFi.h>
    #include "./libraries/PubSubClient.h"

    // Digital Temperature Sensor
    #define DHTPIN 2     // what digital pin we're connected to
    #define DHTTYPE DHT22   // DHT 22  (AM2302), AM2321
    DHT dht(DHTPIN, DHTTYPE); // third parameter: 11 for esp8266-01

    // AP wifi
    const char* apName = "hello-world";
    // MQTT
    const char* mqtt_server = "192.168.1.111";
    int mqtt_port = 1883;
    const char* clientUsername = "pippo";
    const char* clientPassword = "ciao123456";
    const char* willTopic = "...";
    const char* willMessage = "...";
    WiFiClient espClient;
    PubSubClient client(espClient);
    long lastMsg = 0;
    char msg[256];

    void message_arrived(char* topic, byte* payload, unsigned int length) {
      Serial.print("Message arrived [");
      Serial.print(topic);
      Serial.print("] ");

      String mex = "";
      for (int i = 0; i < length; i++) {
        mex += (char)payload[i]; //Serial.print((char)payload[i]);
      }
      Serial.print(" - Messaggio: ");
      Serial.print(mex);

    }

    void setup() {
      // setup once:
      Serial.begin(115200);
      WiFiManager wifi;

      wifi.setTimeout(120); // retry after 2 minutes to connect to a wifi.

      if(!wifi.autoConnect(apName)) { // timeut reached
        Serial.println("failed to connect and hit timeout");
        delay(3000);
        //reset and try again
        ESP.reset();
        delay(5000);
      } 

      // if you get here you have connected to the WiFi
      Serial.println("connected!");

      client.setServer(mqtt_server, mqtt_port);
      client.setCallback(message_arrived);
      
      dht.begin();
      
    }

The last lines raises up this error:

    sketch\mqtt-temperature-wifi-manager.ino.cpp.o: In function `WiFiManager::~WiFiManager()':
    
    C:\Users\Rocco\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:23: undefined reference to `PubSubClient::setServer(char const*, unsigned short)'
    
    C:\Users\Rocco\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:23: undefined reference to `PubSubClient::setCallback(void (*)(char*, unsigned char*, unsigned int))'
    
    C:\Users\Rocco\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:23: undefined reference to `DHT::begin()'

Am I wrong importing the libraries or is it a WifiManager issue?

roccomuso, 2015-12-08:
My fault, i was badly importing the libraries.

closed 2015-12-08 by roccomuso

ivanadiguna, 2018-01-18:
I got this error too, which libraries u false importing?
thx