tzapu/WiFiManager #21: Callback when in AP mode

alwynallan, 2015-12-16:
Others might have this need, so feel free to integrate my solution. My application has a display but no user input. It's useful to display that the device is in AP mode, waiting for configuration. I did it with a callback function: in WiFiManager.h, public:

`void    setAPCallback( void (*func)(void) );`

and private:

`void (*_apcallback)(void) = NULL;`

Then in WiFiManger.cpp after `WiFi.mode(WIFI_AP);` I added

`if( _apcallback != NULL) _apcallback();`

and the obvious

`void WiFiManager::setAPCallback( void (*func)(void) ) {`
`  _apcallback = func;`
`}`

Feel free to improve. Thanks for the handy library.

tzapu, 2015-12-16:
Hi,

this is very nice, i will add it.

something like this i plan to also add for additional variables, thank you very much, you hva kickstarted me in the right direction.

cheers
alex

tzapu, 2015-12-16:
I will also try to add a display/oled example for this, maybe i ll mange tomorrow
thank again

tzapu, 2015-12-16:
probably worth sending the ip of the AP in the call back as well

domonetic, 2015-12-16:
I think one would overload of unnecessary code, there are functions to find the ip.

tablatronix, 2015-12-16:
Got a pull request or commit ? I was about to do this myself.

tzapu, 2015-12-17:
@domonetic you are right, makes sense
@tablatronix  i plan on merging it today, as soon as i get a chance. will update here

tzapu, 2015-12-17:
guys, pushed an update, will update documentation as well later
thanks again

tzapu, 2015-12-28:
this is done and seems to work fine

closed 2015-12-28 by tzapu

alwynallan, 2015-12-28:
Thanks guys for including this. I have upgraded to your version and it all works. One day I'll learn git fork/merge and make my changes myself.