tzapu/WiFiManager #22: public and private methods

domonetic, 2015-12-16:
I am reviewing the code to help with the wiki and find something I do not understand.
begin () is called by AutoConnect () but is declared public and I think it should be private.

tzapu, 2015-12-17:
yes, you are correct again, leftovers from a past version. 
it allwed me then to have an oled connected and do all the steps in the app rather than the library so i can display updates on the screen.
no longer possible, code cleanup is on the table, together with the new callabacks

so nice to see so many people taking interest, thanks

domonetic, 2015-12-17:
These have no implementation. It is a virtual method? if I remove works just

**String beginConfigMode(void);
void startWebConfig();
String  beginConfigMode(void);
int serverLoop();**
is used internally, it should not be publicly exposed
**String urldecode(const char*);**

tzapu, 2015-12-17:
hi, did some cleanup as you indicated.]

urldecode i m gonna leave public, as an utilitarian function for others making url requests

domonetic, 2015-12-17:
Well, I'm helping with the wiki. I hope it will help.

tzapu, 2015-12-19:
thank you very much, just seen it, so nice to have

tzapu, 2016-03-08:
there s so much changed, thining at some point the whole things needs another cleanup, then wiki remade
in the mean time, i ll close this

closed 2016-03-08 by tzapu