tzapu/WiFiManager #24: ap scan speed

tablatronix, 2015-12-21:
I am at a location where I have maybe 25 AP in range with the same name.

Sometimes the list dumps out immediately, and the page pops up.
Sometimes each one ticks by 1 every 10 seconds, and it takes forever, and the page times out.

I know i can skip scans, but why do you think this is? Maybe i can fix it since I can reproduce easily.

tablatronix, 2015-12-21:
I didn't realize there was a `yield` in there, maybe its something in my code.

tzapu, 2015-12-22:
you may wanna try with delay(0) in the library instead of yield... there s been some discussions i ve seen some time ago that it improved things in some library ...

if you see them loading slowly on your screen, it i think it means the data is being sent very slowly. are you far from the esp? Does your firmware have some other loops/timers in parallel ?

cheers

tablatronix, 2015-12-22:
Im sure its another delay blocking. I have a button library scanning for button presses I recently added.
Works fast when I commented out the yield.

( hehe, I think the scheduler library still says beta.)

tzapu, 2015-12-23:
does it work any better with delay(0) instead of the yield()?

tablatronix, 2015-12-23:
It sees to have resolved itself, if it comes up again I will check.
Not sure what changed.

tzapu, 2015-12-27:
could you tell me if you were connecting to the esp from a mobile device or a computer/laptop?

tablatronix, 2015-12-27:
IPHONE with serial monitor running on pc

tzapu, 2016-01-04:
i ll close this. feel free to reopen it if you run into that again
thanks

closed 2016-01-04 by tzapu