tzapu/WiFiManager #25: resets

tablatronix, 2015-12-21:
sometimes when saving I get

*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...

Exception (28):
epc1=0x4000df2f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000073 depc=0x00000000

ctx: cont 
sp: 3fff0990 end: 3fff0ea0 offset: 01a0

tzapu, 2015-12-22:
i used to have this happen, that s why i originally opted to have a reset after every setting.

does it connect ok after reboot or does it forget the settings?

tablatronix, 2015-12-22:
It forgets.

tzapu, 2015-12-27:
apparently there are some issues with wifi connecting and the latest sdk and so on, saw some chatter about this.
what version of the esp8266 package are you using?

tablatronix, 2015-12-27:
git version, it got a bit more stable since i switched, from stable. This sketch is at work, and i am off until the 4th and i don't have access to it, I will work on it more when i return.

tzapu, 2016-01-04:
you might want to update again your github release, there s been a big remake of the wifi library in the arduino package over the holidays. should be a better

tzapu, 2016-01-09:
i will close this, it seems a lot of instability is due to the SDK, 1.5.1 seems to improve things across the board

feel free to reopen if issues persist after updating

closed 2016-01-09 by tzapu