tzapu/WiFiManager #26: Duplicate Wifi networks

lunanigra, 2015-12-22:
Hello, when scanning in our company all Wifi networks are listed 2-3 times.

I didn't had the chance for further investigation; but I assume following reason... We have several access points providing the same Wifi network. Means we can use network XYZ inside the whole building on all floors. So, assumption is that a Wifi network is getting listed multiple times if multiple access points are in range.

Is there a chance putting all items with the same name together?

Thanks, JC

tablatronix, 2016-02-18:
:+1: 
I had this on one of my other issues and it was closed, I forgot to pull it out into its own issues.

This becomes a bit more complicated quickly as there could be some secure some open, different signal levels so you gotta max the best one for display, maybe also show the count (n) so you know it is a institutional AP.

tzapu, 2016-02-19:
it would be easier to do in js... maybe it s worth considering, as then you would only have to pass the array to js, and a progmem string containing the actual js...

what i m mostly worried about is uncontrolable growth and memory usage with this, which is not justified in my mind as you are really only using the lib once in a blue moon to make your life easier, but decreases resources on your ESP at all times...

tzapu, 2016-03-08:
i am close this, don t think there s a big need for it really
if anyone strongly feels different, reopen for another discussion

tablatronix, 2016-03-08:
Im gonna pr it regardless

tzapu, 2016-03-08:
:))
thank you, i ll reopen then

tzapu, 2016-03-09:
@tablatronix Shawn, i just have a thought. 
If you are considering going down the javascript route for this, which it would be a lot easier, there a new function i added to add custom <head> elements. ( `setCustomHeadElement("<script> function doSomething() {}</script>")
it could be then offered as a custom head/script snippet that anyone can add, and whoever is not interested can choose to skip it.
there could be a whole range of customisations offered like this for maximum flexibility

tablatronix, 2016-03-09:
I was gonna just do another n2 loop and remove them but only if there are duplicates detected, but I noticed you changed your sort loop to std::sort comparison

Is it any faster ?

I guess a map might work, I barely know C though and had a hard time trying to modify references in a lambda. shrug.

I was also going to use the sort loop to check if there was any dups before bothering another loop. 
Or do them both in one loop, but it wound up being easier to not combine the two in case the sort algorithm needed to change for memory vs speed on different platforms. I imagine n2 loop is pretty fast even at 16mhz...

Ill look at the JS solution again since you added that callout.

tzapu, 2016-03-09:
welcome to the club, i barely know c as well :)
the std::sort came from a discussion on how to best do it. i doubt very much that at the amount of n s we have, it will make any difference. it was more of a try and see if you can do it kind of thing :d
any way of doing it would probably not have much impact, it might make sense to pull the networks in an array before any looping first, that might make it faster...

js would be interesting, maybe with a count as well, but that would probably waste the most memory as well...

tablatronix, 2016-03-09:
yeah i looked at a hash table and it was not worth even looking at for memory, the extra microsecond for a n2 loop was better, for eg i have 22 duplicate network at this one location, more elsewhere.

tablatronix, 2016-03-09:
Well it looks like std::sort uses quicksort and is around 30% faster in my real world test.

tablatronix, 2016-03-09:
Before
[image]
After
[image]

tablatronix, 2016-03-09:
I added `_removeDupAps` global in case you want to expose it or toggle it.

lunanigra, 2016-03-09:
Cool :-)

tzapu, 2016-03-10:
that s perfect, will pull it in a bit later and add the toggle for it

boy you ve got lots of networks :))

PR #124 (2016-03-16): Remove duplicate aps
fixes #26
commit 27679507 (2016-03-16): add remove duplicate ap feature fixes #26
closed 2016-03-16 by tzapu