tzapu/WiFiManager #27: Storing multiple accounts

lunanigra, 2015-12-22:
It's more an enhancement than an issue ;-)

Currently the latest network is stored in EEPROM. Would it be possible to store up to 3-4 network credentials.?

The idea is that the WifiManager can then loop thru all credentials. Would be cool when moving between some networks, e.g. use an EPS8266 at home and in company as well.

Thanks, JC

domonetic, 2015-12-22:
Wifimanager is to simplify the settings. Currently it stores the data in an area outside the EEPROM.
I think you can use a json file and store it as a file in the area spiffs
trying to connect with autoconnect and if you can not, through the file trying to connect. When connected, the next time Autoconnect do it alone.

domonetic, 2015-12-22:
I adding const char* aNets[][2] to try connect in a list of nets. @tzapu can you check this?

tzapu, 2015-12-22:
i did, i need to think on it a bit more, the patch looks ok so @lunanigra  could use it directly for now

thanks

lunanigra, 2016-01-04:
Happy New Year :-)

Did you have a chance thinking a bit more about multiple account? Or merge the changes into your master?
On my test I ran on some (perhaps small) issues, please see... https://github.com/tzapu/WiFiManager/pull/28

tzapu, 2016-01-05:
happy new year, 
i did think and i will add it, i will refactor it a bit from @domonetic 's solution to look more like WiFiMulti (have a function to add secondary accounts to a list). So if someone wants to use it, they have the ability, while the rest of the library retains it's simplicity.
Regarding your issue, @domonetic was kind enough to provide an update for the .h file as well here https://github.com/tzapu/WiFiManager/pull/29
if you can, try with that for the time being, until this makes it's way in the trunk
thanks

tzapu, 2016-03-08:
this has not been forgotten, i will add it once i m satisfied WiFiManager itself is rock solid

torntrousers, 2016-10-11:
Just a ping about this. I'd find this really useful feature.

kentaylor, 2016-10-12:
Yes, multiple SSID's would be good. Here is a suggestion on how to do it. https://github.com/tzapu/WiFiManager/issues/166#issuecomment-218134202

f0m3, 2016-12-03:
Yes! That would be a cool feature indeed!

closed 2018-01-26 by tablatronix

tavdog, 2019-06-22:
Has this feature been implemented ?

tablatronix, 2019-06-23:
No it requires storing data somewhere