tzapu/WiFiManager #30: Invalid AP password gives no warning

paccerdk, 2015-12-22:
I've spent a few hours trying to figure out why i couldn't change the default AP SSID / password.

The module would come up with the default "AutoConnectAP" and no password,
and seemed to do so regardless of what i did.

Turns out i picked a password which was too short.

Using a password less than 8 characters results in the module using previous saved AP settings.

This is probably intended behavior, but i suggest printing a warning when using passwords <8 characters, or maybe a comment mentioning this.

tzapu, 2015-12-23:
hi, already mentioned here https://github.com/tzapu/WiFiManager#password-protect-the-configuration-access-point

i did not add a warning as i did not know what the criteria was, maybe simply 8 chars or longer is enough as a check

paccerdk, 2015-12-23:
After a bit of digging around, I've come to the following conclusions:

Pass-phrase must consist of 8 to 63 ASCII-encoded characters in the range of 32 to 126 (decimal)

A valid SSID consists of 0-32 octets with arbitrary contents.

This is according to the IEEE 802.11 standard

Source:
http://standards.ieee.org/getieee802/download/802.11-2012.pdf

tzapu, 2015-12-23:
thanks for this
added warning in latest commit

closed 2015-12-23 by tzapu