tzapu/WiFiManager #31: ap doesn't show up, ip reports 0.0.0.0

tablatronix, 2015-12-23:
I read this somewhere, maybe in the source code, I cant remember.

I get a
```
Entered config mode
0.0.0.0
*WM: 
*WM: Configuring access point... 
*WM: ESP457230
*WM: AP IP address: 
*WM: 0.0.0.0
*WM: HTTP server started
```

And my AP is never available to devices.

Do you know what causes this ?

tablatronix, 2015-12-23:
`delay(500); // Without delay I've seen the IP address blank`

hmmm

tzapu, 2015-12-23:
hi, tried it with delay?
never seen it like this in the non modified framework, maybe due to the new callback?

maybe try the simplest autoConnect example

tzapu, 2015-12-23:
hi,

i ve just noticed this happening if i have 
wifiManager.resetSettings(); in code.

do you have that function call in your code?

tablatronix, 2015-12-23:
I added a delay in setup() for debugging something to make sure a pin was low before starting up stuff.

```
setup(){
delay(200);
```

It must be doing some init blocking, which is odd.. But it is probably in the main esp8266 library, so ill close this.

tzapu, 2015-12-23:
it s very odd, i have noticed this happens when i do WiFi.disconnect() and have a delay after.
with no delay after disconnect, everything works fine
quite weird indeed

tablatronix, 2015-12-23:
Yeah and I cannot reproduce yet with a simple test. ( no wifimanager)

tablatronix, 2015-12-23:
With delay(100) it works again, i am gonna guess this is a race condition against the esp8266 libs auto connect.

This says that it will auto connect if there is a stored SDK settings
https://github.com/Links2004

tzapu, 2015-12-23:
well, i got a minimum sketch to replicate this without WiFiManager
better report it to esp8266 arduino
```
#include <ESP8266WiFi.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.disconnect();
  delay(200); //this delay, if you comment out it all works, if left, softAPIP returns 0.0.0.0
  WiFi.mode(WIFI_STA);
  WiFi.begin("", "");
  int connRes = WiFi.waitForConnectResult();
  WiFi.mode(WIFI_AP);
  //assuming connecting failed
  WiFi.softAP("asdfghj");
  delay(500); 
  Serial.println("");
  Serial.println(WiFi.softAPIP());
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

tzapu, 2015-12-23:
tracking https://github.com/esp8266/Arduino/issues/1292

tablatronix, 2015-12-23:
I added a

```
void setup() {
    WiFi.mode(WIFI_OFF);  
    delay(200);
```
to fix for now.

tablatronix, 2015-12-23:
Of course i think that workaround also breaks STA connections, i think this is an issue with dhcp client and servers starting up, either way someone smarted can look at the issue i created and figure it out.

Thanks

tzapu, 2016-01-09:
this seems sorted with the latest github changes and sdk 1.5.1

closed 2016-01-20 by tzapu