tzapu/WiFiManager #32: refactor: abstract ap scanning

tablatronix, 2015-12-23:
Abstract wifi scanning out of `handleWifi` so it can be called on its own.

Probably need to abstract most of the html outputters so they can be called internally or used to callout to ` server->sendContent(item);` externally.

make the code more portable

tzapu, 2016-03-08:
well, do you think there would still be a need for this?
there isn t much you can do realy with scanning...
also, most users of this lib, i don t think would mess about with html through abstractors, etc

currently to me it just seems a lot of effort with little to no payback

tablatronix, 2016-03-08:
I thought you had plans to add templating

tablatronix, 2017-08-28:
abstracting scanning needs to be done to allow live ajax scanning updating and prescan so the config page does not have to wait on it