tzapu/WiFiManager #33: Add singal strength in AP listings

tablatronix, 2015-12-23:
Add quality as a standard range or -50-0dbm.
ideas ?

```
int getRSSIasQuality(int RSSI){
  //  int dBm = WiFi.RSSI();
  int quality = 0;
  
  if(dBm <= -100){
        quality = 0;
  }else if(dBm >= -50){
        quality = 100;
  } else {
        quality = 2 * (dBm + 100); 
  }
  return quality;
}
```

Any easy way to sort this also, client or server ?

tablatronix, 2015-12-23:
Also show duplicate ap names as one and auto select best ?

tzapu, 2015-12-25:
nice, will add the rssi indicator/quality
will think of the duplicate rssis, i think i need to get some hardware setup to test this
cheers

tablatronix, 2015-12-25:
- [ ] add strength filtering optional threshold

tablatronix, 2015-12-25:
I figure it can act like most computer selectors, not reinventing wheel , we can probably find the specs everyone uses. I know they remove low signals from listings, and they combine duplicate and and auto select somehow. Right now my module shows 3 times as many APs as my other devices do, obviously we dont' need qos for IOT, so this might be a bit different, but might make auto selecting best easier. and simplifying the interface.

tzapu, 2015-12-25:
maybe a flag to not show all networks with quality 0
no reason why that can t be an arbitrary value though

btw, i am seeing more ap s on the esp modules as well, but, i can also connect from further away from my esps than my phone for instance, they just seem to outperform a lot of other wifi devices

tzapu, 2016-01-11:
added filter for low quality signal set by using   `wifiManager.setMinimumSignalQuality();`
will need to add documentation soon, it does take a parameter that sets under what percent to ignore networks, defaults to 8%
inspired from http://www.speedguide.net/faq/how-does-rssi-dbm-relate-to-signal-quality-percent-439

domonetic, 2016-01-11:
Back my holiday documented all.

tzapu, 2016-01-13:
this is done, documentation will be coming shortly, there is another issue in which we can discuss dropdowns and duplicate networks
cheers

closed 2016-01-13 by tzapu