tzapu/WiFiManager #34: Auto join open APs

tablatronix, 2015-12-25:
Can esp8266 lib even detect this ?

Combined with strength logic, this might be neat for a roaming option.
Could also add wildcard ap matching, for example how xfinity and atts free wifi works.

tzapu, 2015-12-25:
i think you can detext open ip networks, but then it needs logic to retry the connection when you fail, and so on.
i am thinking that might be better off in a separate library just for that, outside the scope of WiFiManager

tablatronix, 2015-12-25:
Maybe we can just add secure flags to the listings for manual ID.

tzapu, 2015-12-25:
yup, will add something like that as soon as i get another chunk of time i can allocate to this

tablatronix, 2015-12-25:
Ill brush up on my C and take a look if i get time.

tzapu, 2015-12-27:
added a padlock indicator for secured networks
please test

closed 2015-12-27 by tzapu