tzapu/WiFiManager #35: Failed connect to new AP after receiving Wi-Fi credentials

uetchy, 2015-12-26:
Failed attempt to connect to new AP after receiving user input Wi-Fi credentials.

```console
*WM: Handle root
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 4
*WM: Failed to connect.
```

After I reset the Arduino board, WM seems loading credentials properly.

tzapu, 2015-12-27:
there might currently be some issues with the esp8266 package and weird behaviour of wifi connecting, maybe this is what affects you as well.
i ll think of re-adding an optional reset to the flow, see if it makes things better

tzapu, 2015-12-28:
any chance you could  try getting the github release of the esp8266 arduino package and try flashing with that?

tzapu, 2016-01-04:
using some new changes in the latest Github head of the esp package, i could add an optional reset after connecting to troubleshoot/fix cases like this.
the downside is that it would mean anyone that wants to use the library would need to be on the latest github release, rather than the current stable release...

uetchy, 2016-01-06:
Thanks.
After pulling latest HEAD of ESP8266 and WiFiManager, this issue seems fixed.

```console
*WM: AutoConnect
*WM: Reading SSID
*WM: SSID: 
*WM: [omitted]
*WM: Reading Password
*WM: Password: [omitted]
*WM: Connecting as wifi client...
```

However, WM is stuck in `WiFiManager::connectWifi`...

tzapu, 2016-01-06:
do you happen to have another ESP8266 module you could try ? or with a better power source?

uetchy, 2016-01-10:
I'm using USB powered ESP-WROOM-02 development board.
I figured out this is some problem of `WiFi.begin`. Anyway, this is not related to WiFiManager so you can close the issue.

tzapu, 2016-01-11:
cheers
i d like to mention there are a lot of wifi changes and fixes in the latest github  arduino esp8266 core

closed 2016-01-11 by tzapu

tzapu, 2016-01-11:
btw, you should try again with the latest github based on sdk 1.5.1
it solved quite a few really random wifi problems...

tzapu, 2016-01-14:
@uetchy i've noticed the same behaviour as you mentioned above with a trendnet router
maybe it helps you in anyway