tzapu/WiFiManager #36: Problem with MF-RC522

VjSky, 2015-12-27:
Does wifimanager interfeere with spi timings? I had a working code that reads rfid cards and after adding wifimanager i get  timeouts when looking for a new card present ( the library returns false, but i tracked internaly that it gives a timeout status ) 
PS: great work!

tzapu, 2015-12-27:
hi, it really shouldn.t
after it s job is done, and after setup() is complete, the library should be even removed from memory and no longer be active at all.
there are currently some weird issues with the esp8266 package and delays and wifi not connecting, also with sdk 1.5 and wifi not connecting. they are all reported, but i don t know if fixing those will fix your issues as well...

VjSky, 2015-12-27:
Thanks for a fast reply!
Does it get removed automatically or do I have to do something to remove it from memory?  Everything else works fine for now.

tzapu, 2015-12-27:
no probs, 
if you followed the examples code, and defined everything locally to the setup() function, it should det deallocated as soon as setup ends. so no global declarations of WiFiManger or the likes

VjSky, 2015-12-27:
I had to check, but I did everything in setup. I'll have to try to figure it out.

VjSky, 2015-12-27:
Got it. 
For some reason, it does not work if you call spi.begin() first, and then use wifimanager. So always be sure to do all the wifi stuff first and then everything else.

closed 2015-12-27 by VjSky