tzapu/WiFiManager #37: windows laptop failed to acquire ip after connecting.

piux, 2015-12-27:
hi, im trying the simple example from  wifi manager and when i connect with my android device, it has no problem and gets the ip 192.168.4.3, can connect to 192.168.4.1 and change the settings. but when i try this with my windows laptop it gives me this ip settings:

ip 164.254.145.161
subnet 255.255.0.0
default gateway: none

it wont connect to 192.168.4.1 unless i manually set a static ip like this:

ip 192.168.4.4
subnet 255.255.255.0
default gateway: 192.168.4.1

piux, 2015-12-28:
nevermind my module isnt working properly, flashed a spare one and it worked fine.

closed 2015-12-28 by piux

tzapu, 2015-12-28:
really weird that the phone would be ok and laptop wouldn't...
i have noticed my laptop is very slow to connect while phones are completely ok... but it always got the ip

piux, 2015-12-28:
@tzapu its the module, isnt working properly, i tested it in another module esp8266 i had, it works fine.