tzapu/WiFiManager #42: add library.json and others files

domonetic, 2015-12-30:
I added the files to standardize the library.
It allows that in a new update, can be made from the Library Manager.

tzapu, 2016-01-02:
thank you very much,
i ll be back in full working order on the 4th and will go through everything asap

tzapu, 2016-01-04:
@domonetic merged the library changes
thank you very much

closed 2016-01-04 by tzapu

domonetic, 2016-01-04:
You are welcome. I took the audacity to propose to be included in the Library Manager.
Please go and approves the inclusion

https://github.com/arduino/Arduino/issues/4364#issuecomment-168278446

cmaglie, 2016-01-04:
@tzapu remember also to move the v1.4 tag!

tzapu, 2016-01-04:
done, thank you again

cmaglie, 2016-01-04:
> @tzapu remember also to move the v1.4 tag!

ops, I mean "v0.4", basically the library.properties file must appear here:

https://github.com/tzapu/WiFiManager/tree/v0.4

domonetic, 2016-01-04:
create a new release.

tzapu, 2016-01-04:
hi, created a new tag, 0.5 and updated the library.json and library.properties files
is it ok?

cmaglie, 2016-01-04:
Well not yet :-) it seems that you created the new tag **before** changing library.properties, so the tag 0.5 now points to a library.properties with "version=0.4" written in it.

https://github.com/tzapu/WiFiManager/tree/0.5
https://github.com/tzapu/WiFiManager/blob/0.5/library.properties

to fix this:
- remove the tag 0.5
- remake the tag 0.5 on master branch.

domonetic, 2016-01-04:
need change version on library.properties, and merge changes in keywords.txt. Please

tzapu, 2016-01-04:
ok, updated everything inclusing keywords.txt ( @domonetic somehow the pull request was merged but the keywords still cut off, modified it directly)
thanks

domonetic, 2016-01-04:
Congratulations, now available in the library manager
https://github.com/arduino/Arduino/issues/4364#issuecomment-168714767

tzapu, 2016-01-05:
thank you very much Maximiliano , you did a great job of everything