tzapu/WiFiManager #43: A better UI with Bootstrap?

teomaragakis, 2015-12-31:
I could work on a much better UI by including the Bootstrap library from the CDN, if you are okay with this.

walthercarsten, 2015-12-31:
Would be great!

tzapu, 2016-01-02:
hi guys, that would be really great, and would look cool and it would take up no ram or flash whatsoever, 
unfortunately, this library is only up and running when the esp is not connected to the router/internet, and when the device you are using to configure with is connected to the esp, so there is no way for the webpage to load the actual bootstrap css from cdn, because there is no internet access

teomaragakis, 2016-01-02:
You know, this fact escaped me completely. I could still try to work on the UI a bit though. I could use this thread to publish drafts.

tzapu, 2016-01-04:
feel free to do so, i would appreciate any input. my graphical skills are rather limited.
do take into account that I am trying to keep the size of everything included to an absolute minimum from a flash/memory usage point of view. which complicates getting something looking nice immensely :P

you really wouldn't want to have to load a huge library in your projects just for the 1-2-3 times that you need to actually configure a module

tzapu, 2016-01-04:
one of the points on my list of todos is to maybe allow the user sending in css, so then you would have full ability of customising it for your own project

teomaragakis, 2016-01-04:
I absolutely agree with the keeping it light part. What do you mean with "allow the user sending in css"? How exactly?

tzapu, 2016-01-04:
sending a variable containing the css to the library, and it will get applied instead of the default one

teomaragakis, 2016-01-04:
I wouldn't waste time (or memory) on that but it's your project :) I believe that a simple good UI for all will work fine and then users should be able to fork to further customize.

tzapu, 2016-01-04:
cool, thanks
any suggestions drafts you might have on how to improve it are more than welcome
cheers

closed 2016-02-04 by tzapu