tzapu/WiFiManager #46: Newbie

yomasa, 2016-01-04:
Sorry for posting this here, But I don't fully understand this yet; I got the example working fine but once
the you get connected how to I use it to start a webserver, Do I refer to the WIFIManager object for the connection?
can someone point me to some examples?

Thanks

VjSky, 2016-01-05:
WifiManager only deals with connecting to the WiFi network. Once connected
it is up to you to implement any desired behavior.

yomasa, 2016-01-05:
Yes but is it still connected? do I use it to do the http read/writes or do I just get the ssid and password from it and do my own thing? reopen wifi connection.... ?

domonetic, 2016-01-05:
This help to you to find,connect,save and remember password and ssid. .nothing more.
After this you coding anything you need

pieman64, 2016-01-10:
Just following on from this I notice there is a WiFiManagerTemplate.html in the extras directory. Is this used in any of the examples or would we need to code this ourselves?
Could someone also explain briefly what each of the examples do. Specifically what is the difference between AutoConnect and On DemandConfigPortal?
Which example should I pick for providing the ESP with the router details via the Android ESP8266 SmartConfig app but with a fallback to configure via a regular web browser if an Android device is not available?

riccibitti, 2016-01-10:
AutoConnect connects automatically to the network. It tries very hard; if the last network used for a successful connection is not available (e.g. you moved the device to a new location), then it starts the configuration portal in order to prompt the user for selecting a different network and password.Unless you specify a timeout for it, you can assume that on the next instruction after a call to AutoConnect the network if up and running.

For some applications, it's not desirable to prompt the user any time the network is not available (e.g. you may know in advance that the network is not alwayis available, like for a device that collects data on the road and downloads it once getting back home); other times, you may want to change the configuration even if the current connection is running ok (e.g. you are testing a device in your house, but you want to preset it to work on your friend's home). For all these cases, OnDemandConfigPortal is a way to invoke explicitely the configuration miniserver only when you want it.

pieman64, 2016-01-10:
Thanks riccibitti. I think for test purposes I want to try OnDemandConfigPortal.
The AutoConnect compiled and flashed ok but OnDemandConfigPortal is failing to compile.

wifiManager.setSaveConfigCallback(saveConfigCallback);
gives:
'class WiFiManager' has no member named 'setSaveConfigCallback'
and     
if (!wifiManager.startConfigPortal("OnDemandAP")) {
gives
'class WiFiManager' has no member named 'StartConfigCallback'

I'm using Arduino 1.64. Any ideas what I need to do to get the example to compile?

pieman64, 2016-01-10:
I notice the OnDemandConfig portal states it requires SDK update to at least 1.5.1. Can anyone direct me this SDK?

tzapu, 2016-01-11:
hi, those errors are not thrown by the sdk, but by the lib, it looks like you may have an older version of WiFiManager. If you installed through board manager, only the stuff in the installed examples is available.

if you wish to use the latest stuff, like OnDemand, you need to get the latest github version of the lib and replace the boards manager one.

To get SDK 1.5.1 you need to use the latest github version of the esp8266 core for arduino. The notice to use 1.5.1 only applies to the commented code, if you don't use 1.5.1 (and it should work fine with the 2.0.0 stable release that has SDK 1.3) you might need to uncomment that line if it doesn t work properly.

The html template is there only for people wanting to customise the default html/css in their fork, to have a starting point. It's what i used to build and test the interface, not actively used by the library as it is.

pieman64, 2016-01-11:
Thanks tzapu I'll try and work through your comments.

closed 2016-01-13 by tzapu