tzapu/WiFiManager #49: Move loop from autoConnect to the main loop

fishix10, 2016-01-06:
Hi. I'm new to GitHub and I'm not entirely sure where to put a suggestion. Thanks for a great library!

Just a suggestion for devices that could be used offline. Instead of doing a loop in `autoConnect()` with timeout and such, you could have the users put the server's `handleClient()` calls in their main `loop()`. This way the device can do something while offline.

tzapu, 2016-01-06:
the problem with that would be that then it would need to be both in STA and AP mode and it just makes the setup and possibilities all that much more complicated.
if you set a timeout, WiFiManager releases after the timeout and then you can run offline completly, there s just the small delay when you start.

fishix10, 2016-01-06:
I see. Thanks for a quick reply! I'll give it a try :)

sonu221010, 2016-01-08:
i got following error do not know how to handle

Arduino: 1.6.5 (Windows 8.1), Board: "Adafruit HUZZAH ESP8266, 80 MHz, 115200"

Build options changed, rebuilding all

C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp: In member function 'String WiFiManager::getPassword()':
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:218:18: error: 'class ESP8266WiFiClass' has no member named 'psk'
     _pass = WiFi.psk();//getEEPROMString(32, 64);
                  ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp: In member function 'void WiFiManager::handleRoot()':
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:332:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_SCRIPT);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:333:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_STYLE);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:334:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_HEAD_END);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:344:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_PORTAL_OPTIONS);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:345:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_END);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp: In member function 'void WiFiManager::handleWifi(bool)':
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:361:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_SCRIPT);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:362:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_STYLE);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:363:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_HEAD_END);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:377:38: error: 'FPSTR' was not declared in this scope
         String item = FPSTR(HTTP_ITEM);
                                      ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:395:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_FORM_START);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:403:41: error: 'FPSTR' was not declared in this scope
     String pitem = FPSTR(HTTP_FORM_PARAM);
                                         ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:414:13: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
     server->sendContent_P("<br/>");
             ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:417:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_FORM_END);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:418:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_END);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp: In member function 'void WiFiManager::handleWifiSave()':
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:454:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_SCRIPT);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:455:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_STYLE);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:456:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_HEAD_END);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:458:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_SAVED);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:460:11: error: 'class ESP8266WebServer' has no member named 'sendContent_P'
   server->sendContent_P(HTTP_END);
           ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp: In member function 'boolean WiFiManager::captivePortal()':
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:501:21: error: 'class ESP8266WebServer' has no member named 'hostHeader'
   if (!isIp(server->hostHeader()) ) {
                     ^
C:\Users\sonu\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:503:84: error: 'class WiFiClient' has no member named 'localIP'
     server->sendHeader("Location", String("http://") + toStringIp(server->client().localIP()), true);
                                                                                    ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

sonu221010, 2016-01-08:
there is also error of eeprom.h

tzapu, 2016-01-08:
hi, you need a more up to date package of the esp8266, try either the stable or the github one
see here https://github.com/esp8266/Arduino#contents

sonu221010, 2016-01-08:
thanx bro its worked

tzapu, 2016-01-09:
no probs

closed 2016-01-09 by tzapu