tzapu/WiFiManager #56: Updated lock icon (The small things...)

markaswift, 2016-01-10:
Hi guys,

I've updated the lock icon and specified a size to make it HDPi compatible. Can this be added?

const char HTTP_ITEM_PADLOCK[] PROGMEM = "<img src='data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAQAAAAEACAYAAABccqhmAAAOUUlEQVR42u2dL7QVtxOARyAQiApERcUKBKICUYFARCAqEIgKBCKioqLiiYqKmh4EAoGoQCCeQCAQFYgKxBUIRAUCUYFAICoqEAhExe+XYR+nr/Rx3727OzuzO993zqf4c5LJZjbJJrkisCbONy83v23+3LzbPDzySXPTfNZ8dYKbIx8f+ze3j/6fm82vmp95VxAARC40rzd/aN6XvlP/1fzfDP4pfaK41zxoXmt+4R0QgDXTSf9Wfyh9B5yjo++rjiJ01KCjhc+9AwawZLQD3ZD+7a4dy7tzD/GP5i/Sj1SYOgCcgg6jdUj9XPw7r4VPpR/FnPcONEAUzkk/ZNYFur/Fv5PO4bvmr9KPDM56NwDA3Jxpft180Hwr/h3SU1241GnOFe9GAbBG33Y6xI+6iOfty2aVPkECrAZdANNv6XN9plu6r6VPlEwPYNF86PhvxL9TLVEdKZEIYHHoJzzdgaeLXd6daA1qAtVEyqdECI3OXXVnXvaFPSt1ClW9GxngJHQV+4X4d5IM6n6CL70bHEDRTS2H4t8psql7JvSg0jnvBwDy8p2wwOetfjH4xvtBgFzoSTw9fef98OM/6hFmDiCBOXpAh0W+mOpnw6veDwisE/0WreffvR9y3K6uDegnQ3YTwmTokH+tJ/TWqh6uYkoAo2HIv1yZEsBgdAjJkH/5fpgSAOyMflvWIaT3wzvmodfbeHRlXLckfy/9DjodzZQjdVrTHfPisT+7efT3dR++3ujzm/Sf27zrNUY9es26AJyKzhuX9IlPpyfa0T9czqkd2epB14VQ3YGnF3noW3UjyzrvoImMjUPwSfRKLn1zej+op73ddSusdkDdfuz9VtMOpZeb6K68JSyUanJncRD+g745o17UoZ1e3/I6hI/+BtOt0Xrv39MAcfuUmuQ770BBHPRNGnFL7+/SD+2Xeolm1/xJ+pt+vGP5sZrsOVAE7xe9In3m00Skw+mL3oGZGE2yhxLr4tO3wl2EqdGf0IrS+fWse4ZLL3SdJdJlKRr3tSVb2IEoc/6s117pQlyU69JeCwuDqdDGfiX+Q/2MHf9jPtyb6D0i0IXB6AusMAH6wHl/6tN78Je6sGdFJ/2PhXi2i365yJ6QV41meM/PU3plGItO29FNRp6js0fiv7cCDNBG9XrD6ELjgfBg7Yq+hW+J3xeDX7wDANOj80yPh0l3x13wrvxC+Ur8RgPVu/IwHXok1ONtoqcJmVOOQ9dsPEZuOmpjo9AK0BX/uT/36cNzw7viK0OnUHMncV2z4cvAgtE599zHehny26FTgrmPJR96VxqGM/e8fyO8MazREd3cn3Grd6Vhf+ae9/P5aD50XWDOz7msBywM3WQz57z/rneFE6IjrTkXB3U9gAXdhaDXP831YPzgXdnE6IhL5+hztfUt7wrD6ejQf64H4sC7svAeHYHN0d46peTkYGD0jTDXAhE3zcZB213XYOZo9yfelYVPo7fPzPEQ3POuKPwHTQIbmaf92eMRkE7mOVLKan9cdGFQr1GzfgZ0gXntF7csjjlWhPXTE9/5YzPXPgG+/ARCj5BaN/grIesvBd2JaX3Vmy4IfuVdUeiH49a3ztLYy0Pn6dYvhY13JaHfpmnd0HzuWyZz/LZj8a5kdqx/jeZX7wrCYHTnHs/HirGe+zPvXz66h996PeCSdyWzYpnddd5/2buCMAn6U2WMAlaG9dufzT7rwvr0IKOAmdEVWKvG1F+LYei/LnQPv+Xx8EPvCmaiiG02r94VBBP0Nxctp4zcBDUTlrv+nnpXDszQrwK6sGv17LA7cAb0sg+rPf+axbn5Zd1cE7sEoGcEOCdijOWKLvO4HFiuH131rtzasVrNZQ6XhyJ2CeCBd+XWTCe8/WEarEYBuumIE6NGWF74wdw/F5b7SG56V26tWJ36YydXTqx2knJtmAF6HNcqY7OLKydWowBdT/rcu3Jr447YNBbf/XNj9VNj33tXbG1YDde+9a4YuKJ3/TOtDI7uy7dopHfCnv/s6BkBi2frjXfF1oTV9U6PvCsGIbC6SZjj5BNhdbXTde+KQQj0yjeL5+tH74qtBYtrnvXIL/u2QdHzJRZHhX/zrtga6MQmO3PhBxxHv91brDHxi8IjqcLwH+yxmgYU74otnUOZvlF0uMd+bTiObgW3SAD8iOxIXsj0jfLMu1IQEj3PP/Wz9ti7UkvHYnGGrAwnoUd5p37WXnpXaslYbdIo3hWDkFSxed5YCByIxfVNel6bz39wEl+ITQLgqPlAfpDpG4PDP7ANi8NB33hXaqlY7AC8710pCI3FfgB2BA7E4v4/GgO2YfHS4Z7Ageh23akbgw1AsA2Laefv3pVaIlZHgFmQgW1YLDxzNHgAndgkAD7JwDasPj3DnlyR6RvhtXelIDz6idgiAXBH4J4Umb4RNt6VgkXwSqZ/9jrvSi2Nr4UEAD5YJAB+dWpPqkzfCFwBBrtgcUVY8a7U0qgyfSMcelcKFoGOFEkAzlh8j+UWINgFvcpr6meP/Sd7okd2p24EjgHDLuhIcepnr3pXammQAMALEsDE6PVbOgTSPdE6v9JVVj2WO3WQEZfs26O+oX3kofS/MrzoH67Rwutv+r0LEFzEJao3Yh3KAjcS6a/5WBzmQcyojg4O9uqBjljM3xGxv88i9I1WVj/jjYi9YTe0fRcgOIgZvL1jn5wN3etscYU3Ip5s2aVjzsWjAAFBzGSYH7ix+lklRNzutR36pzm3AgQCMaMhLhx9HiAQiBnVvTaunwXPBggCYmYvntpLDekCBAAxs0Ucsbi8ExF39+apvdSQEiAAiJmt4kgJEADEzFZxpAQIAGJmqzhSAgQAMbNVHCkBAoCY2SqOlAABQMxsFUdKgAAgZraKIyVAABAzW8WREiAAiJmt4kgJEADEzFZxpAQIAGJmqzhSAgQAMbNVHCkBAoCY2SqOlAABQMxsFUdKgAAgZraKIyVAABAzW8WREiAAiJmt4kgJEADEzFZxpAQIAGJmqzhSAgQAMbNVHCkBAoCY2SqOlAABQMxsFUdKgAAgZraKIyVAABAzW8WREiAAiJmt4kgJEADEzFZxpAQIAGJmqzhSAgQAMbNVHCkBAoCY2SqOlAABQMxsFUdKgAAgZraKIyVAABAzW8WREiAAiJmt4kgJEADEzFZxpAQIAGJmqzhSAgQAMbNVHCkBAoCY2SqOlAABQMxsFUdKgAAgZraKIyVAABAzW8WREiAAiJmt4kgJEADEzFZxpAQIAGJmqzhSAgQAMbNVHCkBAoCY2SqOlAABQMxsFUdKgAAgZraKIyVAABAzW8WREiAAiJmt4kgJEADEzFZxpAQIAGJmqzhSAgQAMbNVHCkBAoCY2SqOlAABQMxsFUdKgAAgZraKIyVAABAzW8WREiAAiJmt4kgJEADEzFZxpAQIAGJmqzhSAgQAMbNVHCkBAoCY2SqOlAABQMxsFUdKgADgfv7VfNF80jxsbpp/NN8GKBuSANDA1827zStb2vJM82rzfvNNgDIjCQBH+mfzoHl2z3b9rPmzkAiWYN2zbSelBAgAnuwD2b/jf4wmgicB6oIkANzRv5s/TdjGOjW4F6BeSALAHZyy8x/nfoC6IQkAt/jAsK11JLAJUEckAeAJ6oLf2Dn/aXzefBegrkgCwI88mKnNfw5QVyQB4DH1O7/12/8D+mWAz4NxrDO1+4mUAAHAfpPPnLAgGMc6c9v/ixIgALh9h58F1wPUGUkAKP3e/rk5G6DeSAJA6Q/2ePAqQN2RBJDeJ05t/yxA3ZEEkF7LzT/b2ASoO5IA0vvMqe1fBKg7kgDS67EIqLAjMIbVqf3fUwIEAPvNOXPyRYA6IwkAj7w8c7uzDyCOdea2/xclQABw/oXAxwHqjCQAPFLn4+dnavNO+ktHvOuMJAA85o8ztfmdAHVFEgB+pJ7Qu2Dc3l8K14dHsxq3+VZKgADgPz4Xu2PB55ovA9QRSQC4xUOjtn4UoG5IAsAd1CQw1UhA3/wPA9QJSQC4hzodGLsmcEkY9ke3jmzjUZQAAcBP+2ZE23IB6DKsI9p4NCVAAHC7Q+kClB1JADjSoXQByo4kABzpULoAZUcSAI50KF2AsiMJAEc6lC5A2ZEEgCMdSheg7EgCwJEOpQtQdiQB4EiH0gUoO5IAcKRD6QKUHUkAONKhdAHKjiQAHOlQugBlRxIAjnQoXYCyIwkARzqULkDZkQSAIx1KF6DsSALAkQ6lC1B2JAHgSIfSBSg7kgBwpEPpApQdSQA40qF0AcqOJAAc6VC6AGVHEgCOdChdgLIjCQBHOpQuQNmRBIAjHUoXoOxIAsCRDqULUHYkAeBIh9IFKDuSAHCkQ+kClB1JADjSoXQByo4kABzpULoAZUcSAI50KF2AsiMJAEc6lC5A2ZEEgCMdSheg7EgCwJEOpQtQdiQB4EiH0gUoO5IAcKSXBrbt9QBlRxIAjvRgYNveDVB2JAHgSP9ont+zXS80/wpQdiQB4ARummd2bNNz0icN7zIjCQAn9Hnz6inteaP5MkBZkQSARm6at6Xv7OXoAbojfYLwLhuSABBxD6s4UgIEADGzVRwpAQKAmNkqjpQAAUDMbBVHSoAAIGa2iiMlQAAQM1vFkRIgAIiZreJICRAAxMxWcaQECABiZqs4UgIEADGzVRzpAgQAMbNFHDkXIACImf3y1F5qDKfHEH182zx7ehe15U6AQCBm9NEO/dOcywECgZjRGzv0z1nYBAgGYib11qZdb3ky51KAgCBm8vpOPXNGbgUICmIGH+zYJ2fnYYDgIK7ZpxJo6P8xWrDDAEFCXKO/Sb/3Jjw/Nt8FCBjiGvxb+h9qCfvmPwn9cYlHAYKHuGQ3Mvzn3UKgiUB/nkrnLq8CBBQxstpHnjV/khm2+f4fpe8efbAIMawAAAAASUVORK5CYII=' height='16' width='16'/>";

tzapu, 2016-01-11:
that's such a large icon, wastes about ~5k if i did not count wrongly, a far cry from the ~150bytes the other one was. i tried to make it as small as possible, this being an utilitarian lib, i really want to avoid any unnecessary  code to be honest.

 is there any chance you could optimise it quite a bit more than that? it surely does not need to be 256x256 for any kind of hdpi display... do try and get it in the 200bytes region though

cheers

markaswift, 2016-01-12:
My fault, try this on for size:

```
data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAACAAAAAgCAYAAABzenr0AAAQK0lEQVRYCQEgEN/vAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAJycnHA8PD3gJCQm9CAgI6wUFBf8GBgb/CAgI6wkJCb0QEBB2KCgoHAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACsrKxIMDAyZAgIC+QAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AgIC+Q0NDZctLS0SAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGRkZOAUFBeMAAAD/AAAA/wAAAP8CAgL7CAgI0QwMDLkMDAy5BwcH0QICAv0AAAD/AAAA/wAAAP8FBQXjGhoaNgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAZGRk+AwMD9QAAAP8AAAD/BQUF5REREW40NDQSAAAAAAAAAAAAAAAAAAAAADQ0NBQQEBBwBQUF5wAAAP8AAAD/AwMD8xgYGDwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACMjIyAEBATtAAAA/wAAAP8LCwupMTExDgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAMDAwEAsLC60AAAD/AAAA/wQEBOskJCQeAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAoKCrkAAAD/AAAA/w0NDZ9LS0sCAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAEBAQAIMDAyjAAAA/wAAAP8LCwu3AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGxsbRAAAAP8AAAD/BwcH0T8/PwgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA5OTkIBwcH1QAAAP8AAAD/HR0dQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACgoKqwAAAP8AAAD/GhoaRAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGRkZSAAAAP8AAAD/CgoKpwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABZWVkGBAQE8wAAAP8HBwfZWlpaAgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAUlJSAggICN0AAAD/BQUF8V9fXwYAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAiIiIuAAAA/wAAAP8MDAyTAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAsLC5UAAAD/AAAA/ycnJywAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAaGhpQAAAA/wAAAP8ODg5qAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAwMDG4AAAD/AAAA/xQUFEgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGBgYQAYGBq8DAwPPAAAA/wAAAP8DAwPRAAAAtQAAALUAAAC1AAAAtQAAALUAAAC1AAAAtQAAALUAAAC1AAAAtQAAALUAAAC1AAAAtQAAALUAAAC1AAAAtQICAtMAAAD/AAAA/wMDA80FBQWzExMTVgAAAAAAAAAAAAAAAABubm4EBQUF7QAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AgIC+zExMRwAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wMDA/kCAgL5AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/CQkJszQ0NBIzMzMSCQkJtQAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/HBwcOAAAAAAAAAAAGBgYOAAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/EhISXAAAAAAAAAAAEhISXgAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AwMD8QsLCwo4ODgUAwMD8QAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAA41NTUcAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAA41NTUcAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAA41NTUcAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/x8fHyQqKiosAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABqamoUAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wUFBe8FBQXxAAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/y0tLToAAAAAAAAAAABra2sMAgIC/QAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/ywsLDAAAAAAAAAAAAAAAAAADg4OnwAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/AAAA/wAAAP8AAAD/CgoKv2FhYQQAAAAAxXPGoJv7UwgAAAAASUVORK5CYII=
```

tzapu, 2016-01-12:
hi, still seems to be > 5kB ?

markaswift, 2016-01-12:
Acceptable?

```
data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAACAAAAAgCAMAAABEpIrGAAAALVBMVEX///8EBwfBwsLw8PAzNjaCg4NTVVUjJiZDRUUUFxdiZGSho6OSk5Pg4eFydHTCjaf3AAAAZElEQVQ4je2NSw7AIAhEBamKn97/uMXEGBvozkWb9C2Zx4xzWykBhFAeYp9gkLyZE0zIMno9n4g19hmdY39scwqVkOXaxph0ZCXQcqxSpgQpONa59wkRDOL93eAXvimwlbPbwwVAegLS1HGfZAAAAABJRU5ErkJggg==
```

tzapu, 2016-01-12:
huge improvement, will play with it

tzapu, 2016-01-12:
result:
https://www.dropbox.com/s/x4snbio1wivgmo6/Screenshot%202016-01-12%2014.33.43.png?dl=0

don't want to be nitpicking, but if you can export again, we can shave a few bytes of if you size it to be 26*26px
that way it aligns itself properly with the text as well

markaswift, 2016-01-12:
The issue is, we're resizing to exactly @2x so it shows correctly on Retina displays, unless we can have an image of 26x26 and specify the size as 13x13?

tzapu, 2016-01-12:
exactly

this is what works best in my test (alignment and size)

[image]

markaswift, 2016-01-12:
I've tried a resize to 26x26 and it looses clarity, I guess due to scaling of the pixels. I think keeping it at 32x32 would be a better option, also allows for easier locating of icons :-)

I presume we just need to adjust the code to correct the alignment...

tzapu, 2016-01-12:
it s fine then, will update it to this in the next push. 
looks nice with 32x32 size and 13x13 in code

thanks for this, i do enjoy the little things

you are also right about the font, worthwhile change

tzapu, 2016-01-12:
changes added, thanks @markaswift

closed 2016-01-12 by tzapu