tzapu/WiFiManager #58: Not working tonight...

alwynallan, 2016-01-12:
I have the latest git pull of both esp8266com and wifimanger. I'm using the calls in setup() just like the examples and the way they've been for a month. I added the parameter to my .setAPCallback() function, but I have the same problem even if I don't set the callback. When my ESP-12 goes into AP mode I can't get the Configuration page (Android, iOS, Win7) at 192.168.1.4. The serial log looks like this:

```
*WM: 
*WM: AutoConnect
*WM: Reading SSID
*WM: SSID: 
*WM: 
*WM: Reading Password
*WM: Password: 
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: 
*WM: Configuring access point... 
*WM: ClockAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: start loop
*WM: 204 No Response
*WM: 204 No Response
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: 204 No Response
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: 204 No Response
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: 204 No Response
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: 204 No Response
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
```

It repeats as long as something's connected to the AP.

tzapu, 2016-01-13:
funny you should say this, i tested last night with the latest esp8266 github and had the same issue.
i had to revert to the 5f80ad5a78062882ff2bf509bae7d69996a53ca7 comit to make it work...

MrBill1234, 2016-01-13:
What is the recommended version of esp8266/arduino?  Your web page says:

For now, you need to use a newer staging version of the ESP8266/Arduino package than the stable version released on 23 July 2015

The current 'stable' version is Nov 30 2015.

tzapu, 2016-01-13:
well, it should work fine with the stable 2.0.0 release
it should also work fine with the 2.1.0RC 
the best one was the github version which had sdk 1.5.1 (newest) embeded, but that broke since last night ...
need to update that documentation to be more explicit, thanks

tzapu, 2016-01-14:
i can replicate the issue you reported when uploading to the latest github
if i revert the changes in e8d60be62a9b08457bed597d721ff44afa57de67 on top of the latest github it works fine.

tzapu, 2016-01-14:
added a fix for this in the latest commit

tzapu, 2016-01-14:
if you can, do test :D

closed 2016-01-14 by tzapu

alwynallan, 2016-01-14:
I tested it and it's back to perfect. Thanks!

It looks good too, and the password is hidden now.

A. Peter Allan · 450 Timber Lane · Devon PA 19333 · 610-687-1708 ·
610-547-4836 mobile

mnelson10000, 2016-01-17:
I'm getting a similar result with the latest version:

*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response

tzapu, 2016-01-18:
hi,

latest release or latest github?

mnelson10000, 2016-01-18:
I started with the latest release, then overwrote the library folder with the latest github,,, both did the same thing.  I went back to the 1/12 commit and it worked, but I had frequent timeouts on the browser.

fmarci, 2016-03-06:
same problem here (from android - from ios it is working)
the library version is 0.10.0

*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP
Entered config mode
192.168.4.1
ESP750091
*WM: 
*WM: Configuring access point... 
*WM: ESP750091
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response