tzapu/WiFiManager #60: Compile error

lovelesh, 2016-01-15:
Hi @tzapu,
I am facing some compile errors while testing the code. The error is somewhat like this
error: 'class ESP8266WiFiClass' has no member named 'psk'

I have checked the ESP8266WiFiClass and it doesnot contain and psk member function.
I am on 1.6.5-947-g39819f0 version of esp8266-arduino.

I am working on similar thing and I have used the apboot code as reference. I am adding extra functionality like GPIO toggling for my RGB bulb.

closed 2016-01-15 by lovelesh

tzapu, 2016-01-15:
hi, as you must have realised because you closed this, you need to be on 2.0.0 or newer for the more recent versions of WiFiManager
cheers

lovelesh, 2016-01-15:
Sorry for not reading the README. I am waiting for you next release as I
want some other data to be saved in PROGMEM. Can you provide any time
frame. I am currently using EEPROM and I find it unreliable. I am looking
for better approach as I am developing a product that should work for
atleast a year.

Looking forward for your suggestions.

tzapu, 2016-01-15:
hi,

you have an example in the current version that saves to file system, in json. seems quite reliable

give it a shot

lovelesh, 2016-01-15:
Thanks tzapu,

I think this code will perfectly suit my requirements. Is there a way I can
send the config.json file directly in SPIFFS without interfering with the
code. I was thinking of providing 2 files 1) default and 2) regular. Can I
push those into the file system before uploading the code?

tzapu, 2016-01-15:
i think you can upload files to the esp beforehand, although i never did it
look in the esp8266 for arduino documentation

p.s. the example takes defaults, so it automatically creates a file for you with the defaults after the first use. you don t necessarily need the file beforehand, just set the appropiate defaults

lovelesh, 2016-01-15:
If I am not able to send the file then I'll settle for automatic file
creation from the code. Right now I'm doing similar thing using global
variables.

tzapu, 2016-01-16:
hi,

see here, uploading files

https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md#uploading-files-to-file-system