tzapu/WiFiManager #61: Intermittent Wifi.begin() on ESP.reboot();

cancodr, 2016-01-15:
First off: awesome work on this library!  It is exactly what I'm looking for.

I noticed a weird issue though.  I am experimenting with the ESP8266 (Adafruit Huzzah Feather to be specific).  In my experiment, I created a console application, so I could interact with it via the serial port.  I'm using WifiManager as the connection framework for Wifi.  I added a "command line" ability to software reboot the module using ESP.reset();

When I use this method, often times the autoConnect() call fails.  That is: Wifi.begin returns 6 but Wifi.waitForConnectResult() starts to return "1", and continues to do so even if I try to run the Wifi.begin / waitForConnectResult() within a loop.

In this case, unless I actually force a disconnect using WiFi.disconnect(), then after a hard reboot, Wifi.begin() and waitForConnectResult() both start to return zero indefinitely  In this state, even a hard reboot (using the reset button) in some cases doesn't correct the problem, and I need to re-upload the sketch.

The funny part is though, if I comment out autoConnect() and manually connected via Wifi.begin() inside the setup() routine, it always works, regardless of  the reboot method.

I have read that ESP.reset() has its own issues, so perhaps there is something going on there, although once WifiManager fails to connect in this state, it seems that it can't instantiate the WiFi.begin() process (it always returns 0).

I'm happy to provide more info if needed?

cancodr, 2016-01-15:
I have attached an example sketch that demonstrates this.  Upon 1st run, it correctly can't connect, and starts the configuration AP.  I enter the data, and it works.

I can power cycle / hard reboot multiple times after this with no issue.

After executing the "reboot" command in the terminal emulator, the module reboots, and returns a "Connection Result" of "1".  After this, if I reset the board (button or power), the "Connection Result" is always "0", even though the SSID and Password have not changed (and are correctly read back from the ESP8266)

[sketch_jan15a.txt](https://github.com/tzapu/WiFiManager/files/92621/sketch_jan15a.txt)

(sorry, new to the GitHub world, if there is a better way to upload the sketch or code, please let me know!)

tzapu, 2016-01-16:
hi, 

for sketches upload, i gues github gists are good

i can't see anything really wrong with your code, it should really work.
what i would try:
- update to the latest esp8266 arduino core github if you are not on it, or if you are on it, downgrade to the stable package. there are a lot of changes in the core 
- try it with another module if you have one handy
- try it with a better power source, people have seen all kinds of weird issues due to a low quality/low current power source

tzapu, 2016-01-25:
i ve made some changes to the connection code as suggested here and added an example that soft resets if connection is not successful and tries again.
could you give it a try, see if it improves anything for you?

cancodr, 2016-01-26:
Thanks for that!  I have had my hands tied for the last few days, but will give it a try.  As per your comments on some of the other issues, I think there is something fishy going on when the ESP8266 switches modes.  I actually was able to get consistent functionality by doing Wifi.begin() on my own, and then only calling your AP function (startConfigPortal()).  That way, I could get it to consistently connect to the WiFi.

I agree though that there is something strange with when the ESP8266 switches modes.  It looks like by default, it is in STA+AP mode.  In this mode, for some reason, startConfigPortal() would show the AP, but I could never connect.

Seems like the SDK needs a lot of work still.

cancodr, 2016-01-26:
Ok, I think with the latest code, the AutoConnect is more stable.  As noted in issue #71, I think the WiFi.begin() works better without an ssid/password, which seems like its what you are doing now (rather than forcing these).

Initially when I tried it, I was getting the same result as before (sporadic error code "0") on the AutoConnect, but after running through the AP setup again, I can't get it to happen.  Will keep you posted if I see it come up again.

odilonafonso, 2016-01-26:
Probably when we use WiFi.begin(ssid, psk) there is a lot of more code to run.... 
But for to use WiFi.begin() is necessary to check if there is consistents ssid/psk values.
I have achieved better results with WiFi.begin() but when I received return code of 0, simply make a small loop and try again up to three times.
This can be done in two ways:
1.Try make WiFi.begin() connection; on success ... go ahead!
Otherwise, try connecting with ssid / psk;
2.Try get ssid / psk (as currently) and in case of success, make WiFi.begin() call

Another thing I have tested...
The tests I've done with the AT firmware (recording the sketch on a Arduino board) are giving good results, but the library is very outdated, it is not getting the same attention as the library (this one) for the use of ESP8266 without Arduino. 
To use the ESP8266 just like wifi module, I have used the library https://github.com/itead/ITEADLIB_Arduino_WeeESP8266.

There is a vastly different development branch of the master, I am in doubt about what I use.
Sorry to be posting this type of question here, but as I have done tests in both forms of work - ESP8266 with and without Arduino - I found no other place to express such doubts.

tzapu, 2016-01-27:
well, most people here will be in favour of using the esp instead of esp+arduino, me included.
it s faster, cheaper, just as easy to develop nowadays, fantastic community... i love it :), then again, i never got involved with simply arduino in this way, or not at all, so who knows

regarding the WiFi.begin(parameter) i can say i ve noticed this:
- it works better
- it s faster
- seems to remember a prev set static ip, so i could build a sketch now that doesn t need to save custom ip parameters anymore
- there is some bug in the esp8266 core or ide somewhere that simply doesn t let the module connect anymore once you call it after a reboot with a set ssid and pass. it is very weird, i ve had it happen to me for half a day yesterday and i was not even using WiFiManager, just begin. 
It only stopped happening after i set the proper flash size to the module from the ide dropdown, but it may have been just a coincidence... maybe i need to retry replicating and reporting it. i m sure it baffled a lot of people, as working and simple code would no longer work....

@odilonafonso are you still getting connection result 0 with just WiFi.begin() ?

odilonafonso, 2016-01-28:
Hi @tzapu,
I am currently involved in a number of projects, I will soon resume testing with the ESP8266.
I remember vaguely when I use WiFi.begin () without the parameters of connection, the most common is to receive 0 only on the first call, then immediately return 3.
Grateful for all the support !!

tzapu, 2016-03-08:
hi, i ve just made another commit to this, could you please try the github version of WiFiManager and see if anything changed for you?

closed 2018-01-26 by tablatronix