tzapu/WiFiManager #62: Static IP when in Station Mode

jonathanendersby, 2016-01-16:
Is there any way to specify a static IP when in station mode?

tzapu, 2016-01-16:
not currently,
on the road map to sort

tzapu, 2016-01-17:
thanks for the pull request
now added, together with suggested modifications to variables
please test and reopen if problematic

closed 2016-01-17 by tzapu

josep112, 2016-01-23:
Hello after update wifimanager to use wifiManager.setSTAStaticIP Config now appears this error:

/Users/joseantoniosanchez/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp: In member function 'void WiFiManager::handleWifiSave()':
/Users/joseantoniosanchez/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:462:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_ip.fromString(server->arg("ip"));
                    ^
/Users/joseantoniosanchez/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:467:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_gw.fromString(server->arg("gw"));
                    ^
/Users/joseantoniosanchez/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:472:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_sn.fromString(server->arg("sn"));
                    ^
exit status 1
Erro compilando.

Pleas help me

Thanks

tzapu, 2016-01-24:
hi, you need to run the latest github version of the esp8266 core for arduino ide, until they release a new staging at least