tzapu/WiFiManager #63: How to assign blynk_token

mocheffendi, 2016-01-16:
i need help how to read blynk_token from portal and assign to blynk.config?

thanks

ME

tzapu, 2016-01-16:
hi, look in the FS example 
https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino

```
 blynk_token = custom_blynk_token.getValue();
```

mocheffendi, 2016-01-17:
thanks for your fast response

blynk_token = custom_blynk_token.getValue();

  Blynk.config(blynk_token);
  //Blynk.begin(blynk_token);
  Serial.println(blynk_token);

if I use Blynk config no error on compile but esp is not on network
but if i use Blynk begin, it cant compile

mocheffendi, 2016-01-17:
'''
#include <FS.h>                   //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <BlynkSimpleEsp8266.h>

//define your default values here, if there are different values in config.json, they are overwritten.
//const char *mqtt_server = NULL;
//const char *mqtt_port = "8080";
const char *blynk_token = "BLYNK_TOKEN";

char auth[] = "419f17ecfe0a45ba88580f6977cb0388";
char authesp[] = "419f17ecfe0a45ba88580f6977cb0388";
//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        StaticJsonBuffer<200> jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          //mqtt_server = json["mqtt_server"];
          //mqtt_port = json["mqtt_port"];
          blynk_token = json["blynk_token"];
        } else {
          Serial.println("failed to load json config");
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  //end read

  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  //WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  //WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 8);
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //add all your parameters here
  //wifiManager.addParameter(&custom_mqtt_server);
  //wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_blynk_token);

  //reset settings - for testing
  wifiManager.resetSettings();

  //set custom ip for portal
  wifiManager.setAPConfig(IPAddress(10,1,1,1), IPAddress(10,1,1,1), IPAddress(255,255,255,0));
  
  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  wifiManager.setMinimumSignalQuality();
  
  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  //wifiManager.setTimeout(120);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("MEfEvaFaraAzzahra")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //read updated parameters
  //mqtt_server = custom_mqtt_server.getValue();
  //mqtt_port = custom_mqtt_port.getValue();
  blynk_token = custom_blynk_token.getValue();

  //char temp[strlen(blynk_token) + 1];
  //strcpy(temp, blynk_token);
  //char * cstr = new char [blynk_token.length()+1];
  //std::strcpy (cstr, blynk_token.c_str());
  //char * myauth[] = &blynk_token[32];
  String temp = String(blynk_token);
  char myauth[temp.length()];
  temp.toCharArray(myauth, temp.length());
  //Blynk.config(auth);
  //Blynk.begin(myauth);
  Serial.println(myauth);
  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
    StaticJsonBuffer<200> jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    //json["mqtt_server"] = mqtt_server;
    //json["mqtt_port"] = mqtt_port;
    json["blynk_token"] = blynk_token;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.printTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
  }
  
}

void loop() {
  // put your main code here, to run repeatedly:
  
  Blynk.run();
}

BLYNK_WRITE(V31)
{
  WiFiManager wifiManager;
  if (param.asInt()){
    wifiManager.resetSettings();
    ESP.reset();
  }
}
'''

tzapu, 2016-01-18:
i replied on the blynk forum, does that work?

mocheffendi, 2016-01-18:
Dear Tzapu, 

I need to use your parameter blynk token in my sketch but it is not work.

The goal is ESP 12E can be reset and connect to new wifi and new token without upload sketch again and again. 

Please help

tzapu, 2016-01-18:
hi
did you try this
http://community.blynk.cc/t/external-esp8266-configuration/1820/34?u=tzapulica

mocheffendi, 2016-01-18:
Yes I did,

After input SSID and password and blynk auth token at portal, then your library wifi manager use the ssid and password to connect to wifi network, after connected and then I use blynk auth token from portal with syntax like this 
blynk_token = custom_blynk_token.getValue();

And then 
Blynk.config(blynk_token).  But it is not work.

You can see from my sketch below.

```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        StaticJsonBuffer<200> jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          //mqtt_server = json["mqtt_server"];
          //mqtt_port = json["mqtt_port"];
          blynk_token = json["blynk_token"];
        } else {
          Serial.println("failed to load json config");
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  //end read

  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  //WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  //WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 8);
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //add all your parameters here
  //wifiManager.addParameter(&custom_mqtt_server);
  //wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_blynk_token);

  //reset settings - for testing
  wifiManager.resetSettings();

  //set custom ip for portal
  wifiManager.setAPConfig(IPAddress(10,1,1,1), IPAddress(10,1,1,1), IPAddress(255,255,255,0));

  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  wifiManager.setMinimumSignalQuality();

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  //wifiManager.setTimeout(120);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("MEfEvaFaraAzzahra")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //read updated parameters
  //mqtt_server = custom_mqtt_server.getValue();
  //mqtt_port = custom_mqtt_port.getValue();
  blynk_token = custom_blynk_token.getValue();

  
  Blynk.config(blynk_token);
  Serial.println(blynk_token);

  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
    StaticJsonBuffer<200> jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    //json["mqtt_server"] = mqtt_server;
    //json["mqtt_port"] = mqtt_port;
    json["blynk_token"] = blynk_token;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.printTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
  }

}

void loop() {
  // put your main code here, to run repeatedly:

  Blynk.run();
}

```

tzapu, 2016-01-18:
hi, could you give me your serial output as well? cheers

mocheffendi, 2016-01-18:
```
mounting FS...
mounted file system
*WM: Adding parameter
*WM: blynk
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO STRT UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
*WM: 
*WM: AutoConnect
*WM: Reading SSID
*WM: SSID: 
*WM: 
*WM: Reading Password
*WM: Password: 
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: 
*WM: Configuring access point... 
*WM: MEfEvaFaraAzzahra
*WM: Custom IP/GW/Subnet
*WM: AP IP address: 
*WM: 10.1.1.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Scan done
*WM: MEFFENDI
*WM: -59
*WM: Sent config page
*WM: WiFi save
*WM: Parameter
*WM: blynk
*WM: 419f17ecfe0a45ba88580f6977cb0388.
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 4
*WM: Failed to connect.
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Scan done
*WM: MEFFENDI
*WM: -61
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Parameter
*WM: blynk
*WM: 419f17ecfe0a45ba88580f6977cb038.
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
Should save config
connected...yeey :)
419f17ecfe0a45ba88580f6977cb038
saving config
{"blynk_token":"419f17ecfe0a45ba88580f6977cb038"}
```

tzapu, 2016-01-18:
and after, does Blynk.run say connecting or anything?
if you put `Blynk.config("419f17ecfe0a45ba88580f6977cb038");` directly as a test, does it work?

mocheffendi, 2016-01-18:
hi,

there is one char is missing in the last 
must be :  419f17ecfe0a45ba88580f6977cb0388
but actual : 419f17ecfe0a45ba88580f6977cb038

mocheffendi, 2016-01-18:
problem solved :+1: 

WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);

modify the length of char from 32 to 33
WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);

thanks Tzapu, your library is awesome

Best Regards

ME

tzapu, 2016-01-18:
glad you caught it, a commit o posted an hour or so ago did exactly that, set a higher size on char

and thanks for your nice words

closed 2016-01-18 by tzapu