tzapu/WiFiManager #65: Help for send Pushover Message

eppi72, 2016-01-17:
Hi
first many thanks for the wifimanager, i'am very enthusiastic!
I need your support for send pushover message.
I declare first my parameters, this works perfectly, but i dont understand the void loop with my funktion when i press the button. My Arduino compiler send the following error:

sketch_jan17a.ino: In function 'void loop()':
sketch_jan17a:165: error: invalid operands of types 'const char [7]' and 'const char*' to binary 'operator+'
sketch_jan17a:166: error: expected primary-expression before '.' token
sketch_jan17a:167: error: expected unqualified-id before '.' token
sketch_jan17a:168: error: expected unqualified-id before '.' token
sketch_jan17a:169: error: expected unqualified-id before '.' token
sketch_jan17a:170: error: expected unqualified-id before '.' token
sketch_jan17a:171: error: expected unqualified-id before '.' token
sketch_jan17a:172: error: expected unqualified-id before '.' token
sketch_jan17a:173: error: expected unqualified-id before '.' token
sketch_jan17a:174: error: expected unqualified-id before '.' token
sketch_jan17a:176: error: expected unqualified-id before '.' token
invalid operands of types 'const char [7]' and 'const char*' to binary 'operator+'

my ino-file:
#include <FS.h>                   //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#define TRIGGER_PIN 0				// Defintion Dani GPIO0 Button

//define your default values here, if there are different values in config.json, they are overwritten.
const char *mqtt_server = NULL;
const char *mqtt_port = "8080";
const char *blynk_token = "YOUR_BLYNK_TOKEN";
const char *pushover_user = "Your_Pushover_User";
const char *pushover_token = "Your_Pushover_Token";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();
  pinMode(TRIGGER_PIN, INPUT);   // Definition Dani

  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        StaticJsonBuffer<200> jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          mqtt_server = json["mqtt_server"];
          mqtt_port = json["mqtt_port"];
          blynk_token = json["blynk_token"];
		  pushover_user = json["pushover_user"];
		  pushover_token = json["pushover_token"];
        } else {
          Serial.println("failed to load json config");
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  //end read

  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 8);
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);
  WiFiManagerParameter custom_pushover_user("user", "user", pushover_user, 32);
  WiFiManagerParameter custom_pushover_token("token", "token", pushover_token, 32);
  

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //add all your parameters here
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_blynk_token);
  wifiManager.addParameter(&custom_pushover_user);
  wifiManager.addParameter(&custom_pushover_token);

  //reset settings - for testing --Die naechste Zeile wurde von Dani auskommentiert, da sonst der ESP nach jedem Stromausfall in den AP Modus faellt
  //wifiManager.resetSettings();

  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  wifiManager.setMinimumSignalQuality();
  
  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  //wifiManager.setTimeout(120);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("myAP", "myAP")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
  

  //read updated parameters
  mqtt_server = custom_mqtt_server.getValue();
  mqtt_port = custom_mqtt_port.getValue();
  blynk_token = custom_blynk_token.getValue();
  pushover_user = custom_pushover_user.getValue();
  pushover_token = custom_pushover_token.getValue();

  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
    StaticJsonBuffer<200> jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    json["mqtt_server"] = mqtt_server;
    json["mqtt_port"] = mqtt_port;
    json["blynk_token"] = blynk_token;
	json["pushover_user"] = pushover_user;
	json["pushover_token"] = pushover_token;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.printTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
  }
}

void loop() {
  // put your main code here, to run repeatedly:
  if ( digitalRead(TRIGGER_PIN) == LOW ) {
    Serial.println("Sending..");
	String PostData = "token="+pushover_token+"&user="+pushover_user+"&message=testmessage&title=testtitle";
	  if (WiFiManager.connect("api.pushover.net", 80)){
      WiFiManager.println("POST /1/messages.json HTTP/1.1");
      WiFiManager.println("Host: api.pushover.net");
      WiFiManager.println("Connection: close");
      WiFiManager.println("Content-Type: application/x-www-form-urlencoded;");
      WiFiManager.print("Content-Length: ");
      WiFiManager.println(PostData.length());
      WiFiManager.println();
      WiFiManager.println(PostData);   
      delay(100);
      WiFiManager.stop();
	  }
  }
}
Many thanks for your help!!
Cheers Dani

tzapu, 2016-01-17:
Hi Dani,

Could you please format your ino correctly with ``` markdown or include in a gist... it s so hard to follow

cheers

eppi72, 2016-01-17:
Hi, yes sorry und many thanks for your help!
Cheers Dani
```
#include <FS.h>                   //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#define TRIGGER_PIN 0				// Defintion Dani GPIO0 Button

//define your default values here, if there are different values in config.json, they are overwritten.
const char *mqtt_server = NULL;
const char *mqtt_port = "8080";
const char *blynk_token = "YOUR_BLYNK_TOKEN";
const char *pushover_user = "Your_Pushover_User";
const char *pushover_token = "Your_Pushover_Token";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();
  pinMode(TRIGGER_PIN, INPUT);   // Definition Dani

  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        StaticJsonBuffer<200> jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          mqtt_server = json["mqtt_server"];
          mqtt_port = json["mqtt_port"];
          blynk_token = json["blynk_token"];
		  pushover_user = json["pushover_user"];
		  pushover_token = json["pushover_token"];
        } else {
          Serial.println("failed to load json config");
        }
[... 83 lines cut ...]
    json.printTo(configFile);
    configFile.close();
    //end save
  }
}

void loop() {
  // put your main code here, to run repeatedly:
  if ( digitalRead(TRIGGER_PIN) == LOW ) {
    Serial.println("Sending..");
	String PostData = "token="+pushover_token+"&user="+pushover_user+"&message=testmessage&title=testtitle";
	  if (WiFiManager.connect("api.pushover.net", 80)){
      WiFiManager.println("POST /1/messages.json HTTP/1.1");
      WiFiManager.println("Host: api.pushover.net");
      WiFiManager.println("Connection: close");
      WiFiManager.println("Content-Type: application/x-www-form-urlencoded;");
      WiFiManager.print("Content-Length: ");
      WiFiManager.println(PostData.length());
      WiFiManager.println();
      WiFiManager.println(PostData);   
      delay(100);
      WiFiManager.stop();
	  }
  }
}

tzapu, 2016-01-17:
hi, wifimanager just manages the connection and configuration of conenction
you need to be using wificlient to connect to an endpoint
something like
```
 // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
  
  // We now create a URI for the request
  String url = "/input/";
  url += streamId;
  url += "?private_key=";
  url += privateKey;
  url += "&value=";
  url += value;
  
  Serial.print("Requesting URL: ");
  Serial.println(url);
  
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  int timeout = millis() + 5000;
  while (client.available() == 0) {
    if (timeout - millis() < 0) {
      Serial.println(">>> Client Timeout !");
      client.stop();
      return;
    }
  }
  
  // Read all the lines of the reply from server and print them to Serial
  while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
  
  Serial.println();
  Serial.println("closing connection");
```

eppi72, 2016-01-18:
Thank you very much for your hint :+1: 
Cheers

closed 2016-01-18 by eppi72