tzapu/WiFiManager #66: Using POST instead of GET

smixd, 2016-01-17:
First of all: Nice Project! Everyone who starts a Project with the ESP8266 needs this lib! :)

Just a small change:
If you use a Password-Field in the HTML to hide the Password chars,
it also makes sense to use POST in the Form-Method?!
At the Moment the "p" Argument is shown in plaintext in the url, wenn you click save...

tzapu, 2016-01-17:
it would make sense, will see about it

cheers

yomasa, 2016-01-28:
please don't do this. I like seeing the password.

tzapu, 2016-02-04:
i guess it could be a flag to switch between get and post...
it just seems too much hassle currently, so will leave this for later

javl, 2016-03-27:
@yomasa Do you mean you don't want to hide the password in the input field, or don't want to use POST? See discussion in #129

dontsovcmc, 2019-11-28:
@tablatronix  close, it solved

closed 2019-11-28 by tablatronix