tzapu/WiFiManager #67: Manually start config portal.

Hooch180, 2016-01-17:
Hello.
I'm invoking startConfigPortal("Lamp settings") when I detect "setup" button pressed in setup() function.

This works well.

Than I have autoConnect().

If for some reason autoConnect fails to connect it enters config mode. How can I make it so that it tries to connect to this network in a loop without entering configuration portal. In short. I only want to enter configuration portal when I request it. I want autoConnect to retry connection till it succeeds.

How to do it?

Also... what is the best method to restart whole chip/program?

tzapu, 2016-01-18:
HI, see example for this. Once you use startConfigPortal, there is no more need for autoConnect.

tzapu, 2016-01-22:
ESP.reset() is the way to reboot everything
i ll close this now, discuss in the other issues

closed 2016-01-22 by tzapu