tzapu/WiFiManager #68: Not connecting to new WiFi without hard reset.

Hooch180, 2016-01-17:
After calling On Demand Configuration and configuring new wifi settings via configuration portal I get.

*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
*WM: Request redirected to captive portal

WiFi settings are good, because after hard reset module connects to this wifi without any problems.

tzapu, 2016-01-17:
hi, could you update to latest github of esp8266 core for arduino?

odilonafonso, 2016-01-18:
Dear tzu,
When you say to update to latest github of esp8266 core for Arduino, what exactly do you mean by that?
I have Arduino (and ESP8266) installed in three places:
/opt/arduino-1.6.5-r5
/home/odilon/.arduino15
/ home / odilon / Arduino (sketch folder and private libraries)
I do not know exactly where the esp8266 core Arduino must be installed
I am very confused by the amount of directories that both the Arduino as ESP8266 use, as well as the speed with which new updates are made.
Every time I do a new update, go through problems.
Could anyone tell me how best to work with Arduino and esp8266 on Ubuntu Linux 4.15?
I'm seriously considering uninstall ALL and start again.
Is a good idea ?
What release of Arduino should I use - the last? And the ESP8266 - 2.1.0-rc1?
I'm using Arduino 1.6.5 for being compatible with the Eclipse and Arduino plugin for eclipse.
But this is not the most important for me, if I need to update the Arduino IDE for the younger and abandon the use of eclipse.
My application is running 100% ok with the AT firmware when I put the application code (webserver and arduino functions) into the arduino hardware (ProMini, Arduino Uno, Nano, etc..) and I use the ESP8266 ONLY for wifi.
But now I want to use the ESP8266 WITHOUT Arduino... But this is proving to be impossible !!

tzapu, 2016-01-19:
Hi,

Easiest for you to try would be to get arduino 1.6.6, and install through boards manager 2.1.0 rc1 of the esp core and 0.6 of wifimanager 

If you are feeling adventurous, and this should work with 1.6.5, Try this https://github.com/esp8266/Arduino/blob/master/README.md#using-git-version-

Make sure you removed the esp8266  older version you have, there should be a hardware/esp8266 com folder somewhere.

odilonafonso, 2016-01-19:
Ok... now I have Arduino IDE 1.6.7 and ESP8266WiFi git-version.
Things appear a little better, but not 100%.
Thanks for help. WiFiManager is a very nice tool !

tzapu, 2016-01-20:
@Hooch180 could you try the change to the lib that @odilonafonso posted here https://github.com/tzapu/WiFiManager/issues/71#issuecomment-172958415

cheers

madcrow, 2016-01-21:
I am experiencing the exact same problem as @Hooch180 with both the following configurations:
* Arduino 1.6.7 + Arduino core for ESP8266 v2.0.0 + WiFiManager 0.6
* Arduino 1.6.7 + Arduino core for ESP8266 GitHub version 2016-01-21 + WiFiManager GitHub version 2016-01-21

tzapu, 2016-01-22:
@madcrow could you try the change that @odilonafonso made, listed above and see if it s better?
cheers

madcrow, 2016-01-22:
@tzapu I am afraid the solution offered by @odilonafonso won't be of any help in this scenario, because in this case the problem should be somewhere around the  connectWifi() function call inside startConfigPortal() and NOT in autoConnect() .

Remember, this is the scenario, when:
1) The ESP8266 tries to connect to an AP via autoConnect()
2) The AP is unavailable or ESP8266 is not preconfigured for that particular AP -> the connectWifi() inside autoConnect returns something != WL_CONNECTED. (And this part of the library works perfectly well.)
3) startConfigPortal() is called by autoConnect() -> the config portal is initialized, waiting for connections
4) User connects to config portal via browser, selects SSID, sets up credentials, and custom config data -> config portal signals connect=TRUE.
5) The if (connect){} part of startConfigPortal() is being executed now, connection to the newly configured SSID is being attempted:
if (connectWifi(_ssid, _pass) != WL_CONNECTED) //line 154 in WifiManager.cpp
This is where the problem is: If the credentials were correctly configured, this connectWifi() call should return WL_CONNECTED. Instead it always returns 0, i.e WL_IDLE_STATUS.

tzapu, 2016-01-22:
you are correct of course...
and since you already tried 'Arduino 1.6.7 + Arduino core for ESP8266 v2.0.0 + WiFiManager 0.6' i m at a loss as to what to suggest... maybe try printing some messages to serial right before connecting to make sure _ssid & _pass did not get overwritten somehow? (poor memory management for instance..)

madcrow, 2016-01-22:
I've just printed the SSID and pass:
...
DEBUG_WM(F("SSID: "));
DEBUG_WM(_ssid);
DEBUG_WM(F("Pass: "));
 DEBUG_WM(_pass);
if (connectWifi(_ssid, _pass) != WL_CONNECTED) {
...

And they were both accurate.

tzapu, 2016-01-22:
bollox...
if you just put WiFi.begin(ssid,pass) without wifi manager,  does it work ok ?

tzapu, 2016-01-22:
does it work after reset for you as well like for @Hooch180  ?

Hooch180, 2016-01-22:
@tzapu Now I swithed to WiFi.begin and manual conifg portal. I'm no longer using autoConnect. So yes. WiFi.begin works.

tzapu, 2016-01-22:
so it must be that s happening when it goes from STA to AP and tries to connect. 
@madcrow if you still ahve the test sketch
could you try to replace the begin in connectWifi with the following
```
WiFi.setAutoConnect(true); 
WiFi.begin(ssid.c_str(), pass.c_str(), 0, NULL, true);
```

madcrow, 2016-01-22:
Yes, if I reset the module, it connects successfully, just like for @Hooch180.

madcrow, 2016-01-22:
Unfortunatly I don't have access to my sketch and HW right now, because now I am writing from my workplace.
But I will try your suggestion later.

tzapu, 2016-01-22:
ok, 
there s another issue here that behaved similarly and a reboot was sorting it, so maybe i just have to add the feature to reset on end of configuration. wifi gets messed up a few times switching modes...

anyway, it would be really interesting to see the results of your test above, as i can t  replicate the issue, i can t do it myself...

madcrow, 2016-01-22:
Of course.
I'll try it after work, I promise ;)

madcrow, 2016-01-22:
Hmm... it is interesting that you cannot replicate thissue. I have a NodeMCU 1.0 board (ESP12E) and a bare ESP12F module. I tried autoconnect on both, and both behaved the same (i.e. reboot was necessary to successfully connect).

tzapu, 2016-01-22:
i tried it on a wemos d1 mini and i ll try now on a nodemcu 1.0

tzapu, 2016-01-22:
i did try it, it did connect for me. i did notice another bug and commited a fix (you never know)
here s what i did
open example AutoConnectWithFeedback, uploaded it, entered ssid/pass, saved, it connected
i did not add any portal password, or change anything else. maybe you can try this as well, maybe there is something different in your flow than mine...

what i would try:
- run one of the examples directly see if it works as it should
- try the snipped of code above, see if it helps
- update to the latest github with the last fix, see if that helped...

madcrow, 2016-01-22:
Experiment 1 

`WiFi.begin(ssid.c_str(), pass.c_str());`

line replaced by
`WiFi.setAutoConnect(true); `
`WiFi.begin(ssid.c_str(), pass.c_str(), 0, NULL, true);`
inside connectWifi(). 

Result:
`*WM: Connecting as wifi client...`
`*WM: Connection result: `
`*WM: 0`
`*WM: Failed to connect`

madcrow, 2016-01-22:
Experiment 2

Uploaded the AutoConnectWithTimeout example.

Result:
*WM: Connecting as wifi client...
*WM: Connection result:
*WM: 0
*WM: Failed to connect

madcrow, 2016-01-22:
Experiment 3

Updated to the latest GitHub code, and recompiled sketch.

Result:
The same :(

tzapu, 2016-01-22:
oh man... disappointing..

i will try and add the option to reset rather than live connect so it at least works without you having to physically intervene 

p.s. one other thing i do after uploading a firmware is that i reset the module before doing any tests. i have found that most of the times i can t do anything right with the esp when the firmware is loaded by the arduino ide without rebooting... grasping at straws here ... but worth a shot

bretskee, 2016-01-25:
Hi,

I have a very similar issue in that all seems to connect OK until there is a connect fail due to a network issue or the like and the portal is started. After config and a remote reboot, the esp can no longer connect. I have been using wifi manager and had this issue and went back to implementing something very simple using 

WiFi.disconnect();
WiFi.setAutoConnect(true);
WiFi.mode(WIFI_STA);
wiFi.begin(ssis, passwd); 

if it fails, i start an AP using 

WiFi.mode(WIFI_AP);
WiFi.softAP(Name);

I then start a telnet session to set some parameters and do a remote software reset.

I seem to be able to replicate the issue fairly consistently in testing. the procedure is as follows:

1. hard boot and the unit connects reasonably reliably
2. go to my router and take the network down
3. let the esp fail on the network and start the AP and telnet session
4. re-enable the wifi network
5. remote reboot
6. MOST of the time the esp will never again connect to the router. I say most because occasionally it will. Hardware reset fixes this most of the time and it connects. 

i am using 1.6.5-r5 and the latest 2.1.0-rc2. i have been having this issue over a number of versions of both however.

wifi debug reports the following:

Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 0
Auto connect: 1
SSID (6): xxxxxx
Passphrase (10): xxxxxxxxxx
BSSID set: 0

I have also tried any combination of your suggestions above. There seems to be something fishy when changing modes. I could be wrong.

tzapu, 2016-01-25:
i ve always thought there was something fishy when changing mode. the first versions of WiFImanager would always reboot before connecting for that reason.
i am working on bringing that option back... for some modules, it seems the only way, at least for now

bretskee, 2016-01-25:
dont know that will help since the issue persists through a software reset. there must be a subtlety between the hardware and software boot processes. do you think this is something that we will need help from Ivan Grokhotkov??

tzapu, 2016-01-25:
a issue on the main esp8266 core branch with the minumum example (without wifimanager, just base wifi functions) that replicates this can t hurt.
the biggest issue is that if the problem is in the SDK, like a lot are, their hands are tied as well..

tzapu, 2016-01-25:
guys, i ve made some changes to the connection code as suggested here and added an example that soft resets if connection is not successful and tries again.
could you give it a try, see if it improves anything for you?

madcrow, 2016-01-26:
Hey @tzapu, I gave it a try and it seems to work well.

tzapu, 2016-01-27:
nice, thanks for letting me know

closed 2016-02-04 by tzapu

WheresWally, 2016-05-05:
I had issues with the .c_str() function.  
```
WiFi.begin(ssid.c_str(), pass.c_str());
```

I use this instead, and dont forget to trim leading/trailing spaces, they screw with it too
```
char ssss[32] = "";
char pppp[64] = "";
ssid.toCharArray(ssss, ssid.length() + 1);
password.toCharArray(pppp, password.length() + 1);
WiFi.begin(ssss , pppp);
```

MariaGemel, 2017-05-05:
@bretskee 
This one works:

WiFi.disconnect();
WiFi.setAutoConnect(true);
WiFi.mode(WIFI_STA);
wiFi.begin(ssis, passwd);

thank you so much!

RottenVanSuti, 2018-04-09:
Hi @MariaGemel Where do you place those functions? in the setup()?