tzapu/WiFiManager #70: WiFiManagerParameter Length Limitations

knnniggett, 2016-01-17:
I am attempting to use the following sketch, which is based off the FSParm example (using an esp-01): https://gist.github.com/knnniggett/7dec273b7b634e955284

I am using seven parameters, and notice that the last parameter, custom_alarm_desc, has a length of 256.

When I compile and run this sketch, I can successfully enter a long string of text into that field from a web browser. When I hit save, the serial monitor indicates all the parameters have been saved in correct json format.

When I restart the esp-01, the serial monitor indicates the json config was successfully read, and the alarm_desc var contains the correct value. However, when that value is passed to the  WiFiManagerParameter custom_alarm_desc object, subsequent getValue() calls return nothing.

If I reduce the length of this parameter from 256 to 40, everything works as it should.

I can duplicate this same effect when using the AutoConnectWithFSPParamters example sketch.
If I increase the length of the blynk token to 80, everything works. However, if I increase the size of the blynk token to 82, getValue() returns nothing.

If I set the length of the blynk token back down to 80, and then increase the size of the mqtt_port from 8 to 16, then getValue() returns nothing for the blynk token while getValue() continues to return a good value for the mqtt port.

As you can see, I seem to be hitting some kind of boundary or memory limitation. I am hopeful someone could educate me on what that might be and what my options are.

tzapu, 2016-01-18:
well, there s a couple of points in which this might go wrong
1. json buffer is 200, so maybe that overflows? try increasing that...
2. i think i am misallocating or not allocating the memory, or in general, jsut doing something that i m not supposed to be doing with said memory. i will have a new example sketch out today at some point, and it would be good if you could try with that as well.

tzapu, 2016-01-18:
hi, 
i added a new example that has some fixes realated to how it treats variable allocation
could you use this a s a base https://github.com/tzapu/WiFiManager/tree/master/examples/AutoConnectWithFSParametersAndCustomIP

of course you can remove or ignore the ip stuff.
this sample needs the latest esp8266 core github

knnniggett, 2016-01-18:
Thanks for the response.

I'll implement the changes from the referenced example and report back. Monday's are a bit busy for me, so I may not get to this until tomorrow evening.

I had previously tried increasing the read and write json buffers from 200 to 1024, which did not make any noticeable difference. On a side note, the json library has an interesting guide in the their wiki, which shows how to more efficiently allocate the static json buffer via JSON_OBJECT_SIZE and JSON_ARRAY_SIZE macros: https://github.com/bblanchon/ArduinoJson/wiki/Memory-model

This library you've got here is top notch. Have you considered taking donations by linking this project to Bountysource?

tzapu, 2016-01-18:
to be honest, didn t even know bounty source existed.
i am not sure about donations, maybe they will put people off helping with the project...
this is really my first foray into oss, so not sure what to expect

i have updated the example you used as well, to use strcpy instead of using potentially occupied memory, maybe it will help

knnniggett, 2016-01-20:
After pulling the latest master branches of this project, the arduinojson project, and esp6266/arduino, I can report success when using AutoConnectWithFSPParamters, which contains the suggested code changes from commit 823767cb8b85f8e2127638372468a4763950da01 . After increasing the size of blynk_token up to 257, calls to getValue() continued to return the correct value rather than emptyset.

However, when I attempted to switch from the example sketch to my own, I ran into a rather nasty exception when attempting to read a json key which didn't exist (recall the esp-01 still had the json keys from the example sketch in it, rather than the keys from the sketch I wrote).

Turns out this is a known issue, but has been a bit elusive in tracking down the root cause:
https://github.com/bblanchon/ArduinoJson/issues/193

To avoid this, I surrounded each strcopy with a condition that calls containsKey(). Don't attempt to read a value if the key does not exist.  It makes the code look rather ugly, but it works. Up to you if you want to add this to the examples, or perhaps just place a warning that this can occur if you change json key names.

Getting a bit off the initial topic, but this experience does beg the question, how does one clear out stored json values? Clearly an upload doesn't clear it. Would one have to write and upload a sketch that erases the config.json file?

In any case, I think we can call this one resolved. Thank you for your help.

tzapu, 2016-01-20:
hi, thanks for the heads up on the bug. this is getting to be quite a convoluted example :P

to clear for testing i used spiffs.format, which is not desirable to use. depending on your sketch, you could just have a pin/mqtt message/blynk virtual pin/http request that erases the json, or overwrites it, or formats or does whatever you want it to do. should be quite flexible, as you decide how you want to handle it :D

i m gonna leave the issue open for a bit, just in case other people run into it, thanks

bblanchon, 2016-01-23:
Hi,

ArduinoJson behaves differently with `char*` and `String`.
With a `char*`, it just stores the pointer: no duplication.
With a `String`, it copies the string in the `JsonBuffer` and then store this pointer.

This means that these 3 lines, at [AutoConnectWithFSParametersAndCustomIP.ino:163](https://github.com/tzapu/WiFiManager/blob/823767cb8b85f8e2127638372468a4763950da01/examples/AutoConnectWithFSParametersAndCustomIP/AutoConnectWithFSParametersAndCustomIP.ino#L163):

```c++
json["ip"] = WiFi.localIP().toString();
json["gateway"] = WiFi.gatewayIP().toString();
json["subnet"] = WiFi.subnetMask().toString();
```

Are introducing 3 duplications, which are probably going to blow the 200 bytes allocated for the `StaticJsonBuffer`.
You must increase the size of the `StaticJsonBuffer`.

However, since your sketch is designed to be only used on ESP8266, which is a device with "plenty" of memory, you can safely switch to `DynamicJsonBuffer`.

**TL;DR**: replace `StaticJsonBuffer<200>` by `DynamicJsonBuffer`.

tzapu, 2016-01-24:
wow, thanks for this, i will update it

tzapu, 2016-01-24:
@bblanchon sorry to trouble you with one more question:
would doing this
```
 json["ip"] = WiFi.localIP().toString().c_str();
 json["gateway"] = WiFi.gatewayIP().toString().c_str();
 json["subnet"] = WiFi.subnetMask().toString().c_str();
```
make it not copy the string?
sorry for being so awful at memory management :P

bblanchon, 2016-01-24:
This is an excellent question.

You can't do that because `toString()` will returns a `String` object that will be destructed after the call to `c_str()`.
Therefore, the `JsonObject` would store a dangling pointer.

That's why ArduinoJson automatically duplicates `String`.

tzapu, 2016-01-24:
thank you @bblanchon 
much appreciated.

i have updated the examples as per your instructions
alex

closed 2016-02-04 by tzapu