tzapu/WiFiManager #71: AutoConnect does not work all the time

odilonafonso, 2016-01-18:
I used the example of WiFiManager AutoConnect to connect to my wifi network.
The first time he got in the other can not.
Why that ?
I must disconnect, what am I doing wrong?
The following log made by AutoConnect sketch.
I also tried it with a sketch not based on WiFiManager using directly ESP8266WiFi API, but got the exact same problem:
The first time connecting OK, in the other the connection returns error - sometimes 0, sometimes 4.
Grateful.
[AutoConnectLog.txt](https://github.com/tzapu/WiFiManager/files/94592/AutoConnectLog.txt)

r666t, 2016-01-18:
I have the same issue.

tzapu, 2016-01-19:
There are a few potential issues that i ve seen similar to this:
- low quality/ not powerful enough supply
- using sdk 1.5.0 (install 2.0.0 stable or 2.1.9 rc1 or latest github)
- router doesn t like esps - i have on that only connects once a day...
- faulty esp modules

adrianchi, 2016-01-19:
Hello, i have this similar issue since yesterday, previously it was working fine.
I tested the ESP and it seems OK, because i can get connected to WiFi via AT commands.
I have 2 extra things to check.

1.- First the connect time via AT commands is about 3 seconds. But if try the wifiManager it quits instantly witch connect error 0, even with the correct configuration SSI/password. How can i check what is going on? why do not connect? is there an timer wich can be extended to wait a couple of extra seconds?

2.- And second, yestedary i upgraded to Arduino IDE 1.6.7... is that versión verified OK?

odilonafonso, 2016-01-19:
I made a little change on the library;
I check if the there is a ssid saved on ESP. If true, I do `WiFi.begin()` call, without credentials.
Apparently called with no arguments, using existing credentials, it is more efficient.
The code:
```
/*
 *odi begin
 */
    int wifiMode = WiFi.getMode();
    if ( !(wifiMode == WIFI_STA || wifiMode == WIFI_AP_STA) )
        WiFi.mode(WIFI_STA);
    DEBUG_WM(F("\nWiFi.getMode():"));DEBUG_WM(WiFi.getMode());
    if ( !(ssid.equals("")) ){
        DEBUG_WM(F("Trying to connect with saved credentials..."));
        WiFi.begin();
        int connRes;
        for (int i=0; i<11; i++)
            if ( (connRes = WiFi.waitForConnectResult()) == WL_CONNECTED )
                break;
        DEBUG_WM ("Connection result: ");
        DEBUG_WM ( connRes );
        if (connRes == WL_CONNECTED)
            return true;
    }
/*
 *odi end
 */
```
into autoConnect() method from WiFiManager.cpp, after line 113
This works for me.

odilonafonso, 2016-01-19:
I think the ESP module needs to be "awakened".
If I stop access for a few minutes, so I do a hard reset, I get WL_NOT_CONNECTED (4) and then to give a second hard reset, I get WL_CONNECTED (3)
It is possible to set a greater "timeout" to `WiFi.waitForConnectResult()` call ?
Where is the low level functions used by the WiFi modules -
```
wifi_softap_get_config()
wifi_station_get_connect_status()
wifi_get_opmode()
status()
...
```
and others ?
I searched, but apparently these functions are compiled, must belong to some library included in the link stage.
Without source, without doc... we do not know how to access them.
I found some documentation:
https://github.com/CHERTS/esp8266-devkit/tree/master/Espressif
Can someone confirm that the WiFi libraries hear use this espressif libraries ?
There is a lot of documentation there.

tzapu, 2016-01-20:
the esp8266 core for arduino uses the expressif sdk to build upon. version 1.3 i think in stable and 1.5.1 in latest github

i will experiment with your change and see about adding it

btw, i am doing tests all the time with ssid blank (since i am resetting the settings a lot to test), and this didn t really happen to me at all

i won t be able to release a new version (0.7) until they release esp8266 core 2.1.0 or at least a new rc because i added some new function call, so it may be a while

i am on arduino ide 1.6.6, have not tested with 1.6.7 yet

adrianchi, 2016-01-20:
Update, i quit Arduino IDE 1.6.7, and continue testing with ... (By the way... I'am using an ESP8266-01)
- Arduino ide 1.6.6. Esp core 2.0.0. 
wifimanager 0.5.0 - OK
With wifimanager 0.6.0 iam getting the same compile error as this other issue #72. Is something wrong with ESP model?

- Arduino ide 1.6.5. Esp core 2.0.0.
wifimanager 0.5.0 - OK
wifimanager 0.6.0 - OK

BUT, i still have the issue with "Connection result: 0"
I get in wifimanger url and configured again the SSID/Password, and i had "Connection result: 0".
Then i reset the ESP and get "Connection result: 3, connected OK"
The connection process has a wait, because it takes a couple of second to connect. When i get "Connection result: 0" it is displayed at once, no wait.

slubb, 2016-01-23:
I have same issue on ESP-12. When connection fails (for example, cable provider resets modem or executes maintenance network) it will no longer connect. It will display connection result:0 and quickly opens a accesspoint. In debugging I do see my Wifi SSID and username and password it doesn't try to connect at all.
I'm now testing the code from odi to see if this will help.

tzapu, 2016-01-25:
guys, i ve made some changes to the connection code as suggested here and added an example that soft resets if connection is not successful and tries again.
could you give it a try, see if it improves anything for you?

mtnbrit, 2016-02-01:
I found if I have a delay() after serial.begin(), then wifimanager wont connect, this is with latest 2.1.0-rc2 and v0.8.0 of wifimanager. removing delay it works.

tzapu, 2016-02-04:
btw, if you have that delay, does the config portal get an ip or 0.0.0.0 ?

mtnbrit, 2016-02-06:
Hi, no the AP doesn't start, for me it hung after "Using last saved values, should be faster". Just sits there forever, which is bad because it locked up my sensor project for many hours, no watchdog restart or anything. Its running on a NodeMCU v0.9 powered by a 1A macbook USB port, so I would imaging the power supply is adequate.

tzapu, 2016-02-07:
weird, maybe some more of this weirdness related to connecting will be sorted in 1.5.2 whenever they add it to the arduino core...

tzapu, 2016-02-07:
if you can get similar results with a simple sketch with delay and WiFi.begin(...) it may be worth opening a ticket on the arduino core

mtnbrit, 2016-02-13:
The issue appears to be that WiFi.waitForConnectResult() can end up in an infinite loop if it can't connect to the AP. So recommend avoiding this call. I built my own while loop including a timeout.

```
//  int connRes = WiFi.waitForConnectResult();  // this ends up in infinite loop
   while(WiFi.status() == WL_DISCONNECTED && !timeout) {
        delay(200);
	timeout = i++ > 100;
	Serial.print(".");
    }
  if (timeout) {
	  DEBUG_WM ("Connection timed out: ");
	  return 0;
  }
```

tzapu, 2016-02-13:
never seen it locking up before i think..
i ll look at the class itself later

does the loop work ok for you?
now i m thinking that maybe waitForConnectResult quits too fast as well causing some issues others are reporting...

tzapu, 2016-02-16:
i looked at WiFi.waitForConnectResult() and it makes sense to revert to something like your version.
will change it as soon as i can

tzapu, 2016-02-16:
@mtnbrit i have added a new option to WiFiManger to set a connect timeout. it uses a loop similar to yours.
in order to use this you d need to update to the latest github version of the lib and call 
`wifiManager.setConnectTimeout(30);` 
before autoConnect. time in settings. 
if you could give it a try before i release it on lib manager, it would be much appreciated...

mtnbrit, 2016-02-17:
Ive noticed that if wifiManager autoConnect is called when wifi is already connected, it wont reconnect and would normally hang up on waitforConnectResult. Now i have the timeout in there, it times out.

So its up the user to check if wifi.status is connected before calling wifiManager autoConnect, or I could suggest including a WiFi.isConnected() check early on the autoConnect function to return if we are already connected.

tzapu, 2016-02-18:
that s a bit of a weird behaviour as in the main WIFI class, if configuration doesn t change, it just returns..
if(sta_config_equal(current_conf, conf)) {
        DEBUGV("sta config unchanged");
        return status();
}

also the loop checks if connected 
```
 if (status == WL_CONNECTED || status == WL_CONNECT_FAILED) {
        keepConnecting = false;
 }
```
so it should return immediately, as you are already connected...
i think somehow your configuration drives it to run around in circles
i will try to make some simplified test sketches that we could use to report issues to main esp8266 core
cheers

tzapu, 2016-02-18:
you might want to enable debug in the arduino menu and see all the core s debug messages, maybe there s some hints there about what s happening in your case

tzapu, 2016-03-08:
hi, i ve just made another commit to this, could you please try the github version of WiFiManager and see if anything changed for you?

gitgrimbo, 2016-05-11:
My ESP8266 also locks up after these lines are printed, running `AutoConnectWithFeedback` example:

    *WM: 
    *WM: AutoConnect
    *WM: Connecting as wifi client...
    *WM: Using last saved values, should be faster

- Arduino IDE 1.6.7
- ESP8266 2.2.0 (via IDE's Boards Manager)
- WifiManager 0.10.0 (via IDE's Library Manager)

kentaylor, 2016-05-12:
Does it fail every time or just some times? Your messages show that it is failing in the autoconnect function. Try the version of WiFiManager at https://github.com/kentaylor/WiFiManager/  which specifically aims to fix the issue of "AutoConnect does not work all the time" and has some other changes as detailed.

odilonafonso, 2016-06-01:
Hi guys... @tzapu, @adrianchi, @gitgrimbo 

After a few months in other activities, I tested today WM library after updating the SDK to version 2.2.0.

It worked very well!

I can now close this issue?

closed 2016-06-02 by odilonafonso

apicquot, 2016-07-31:
Hello, I am having the same problem with the 2.3.0 version. Has anyone the definitive solution ?
I have played with the WiFi.begin() without attributes, I have replaced WiFi.waitForConnectResult() with my own waiting loop. I have tried to switch the mode to WIFI_OFF and then back to WIFI_STA.

- The device connects fine after a hard reset
- It connects fine in the event of a button reset
- It fails to connect after a deep sleep
- After a few failed connections, the device even crashes with the message "ets Jan  8 2013,rst cause:2, boot mode:(3,6)"

Any help very welcome.

apicquot, 2016-07-31:
Some more information: the error after  WiFi.waitForConnectResult() is WL_NO_SSID_AVAIL. So I tried to scan for networks after a failure, WiFi does not find any networks...

I have tried in Setup():
 `ESP.eraseConfig();
 WiFi.persistent(false);
 WiFi.disconnect();`

It did not solve the problem and curiously WiFi still knew about the nework's ssid, does it mean that the persistence is still on ?

Any help very welcome.

tzapu, 2016-07-31:
hi,

you mentioned deep sleep, are you using RF_DISABLE or any paramenters to the sleep function?

i have found that you can t properly awake the wifi unless you do another deep sleep cycle with RF enabled. but this was some time ago...

cheers

apicquot, 2016-07-31:
Your are absolutely right, I am using `ESP.deepSleep(5000000, WAKE_RF_DEFAULT)` which is the cause of my trouble. Everything works great with `ESP.deepSleep(5000000, WAKE_RF_DEFAULT)`

I assumed wrongly that the rf will be turned on after a reset. Thank you for you help tzapu.

kevcavein, 2018-09-13:
Hi I recently just started testing in esp deepsleep and I found that sometimes my nodemcu wakeup after deep sleep it cannot connect to the WiFi and start the ap. 
It sometimes happen during the second wakeup, sometimes happen after one hour and even sometimes happen after 12hrs. I'm very new to Arduino and I'm not sure what is wrong. 
I'm using the latest wifimanager and from sketch from tzapu samples.
Does Wake_Rf_Default solve the issue?

Sorry if I post on the wrong treat.

tablatronix, 2018-09-13:
This issue is 2 years old open a new ine

EgHubs, 2020-11-27:
> I found if I have a delay() after serial.begin(), then wifimanager wont connect, this is with latest 2.1.0-rc2 and v0.8.0 of wifimanager. removing delay it works.

You saved my life
I've been trying with the default code of AutoConnectWithFSParamtersAndCustomIP for 2 years now and the code sometimes connects and a lot of other tries it doesn't, I was just keeping on pressing reset till it connects, after I deleted the delay line it connects every time just the moment I click on reset.
thank you, I just don't know why that delay line exists!, does it works with others!? I may have tried with more than 5 ESP8266-12 and with 2 NODE-MCU and it does that in all of them, I hope they update the library.

VIet942399, 2021-05-15:
> the esp8266 core for arduino uses the expressif sdk to build upon. version 1.3 i think in stable and 1.5.1 in latest github
>
> i will experiment with your change and see about adding it
>
> btw, i am [...]

> hi,
> 
> you mentioned deep sleep, are you using RF_DISABLE or any paramenters to the sleep function?
> 
> i have found that you can t properly awake the wifi unless you do another deep sleep cycle with RF enabled. but this was some time ago...
> 
> cheers

hello tzapu!!! I thank you for contributing so much to the community!!! i have a problem with your wifimanager library, the problem i have is: initially if i don't have wifi at home then my ESP's hardware codes won't work, because i can't access the loop, because you stay at the wifi connection