tzapu/WiFiManager #72: Errors

yomasa, 2016-01-19:
Error when try to run

WARNING: Category '' in library UIPEthernet is not valid. Setting to 'Uncategorized'

/Users/jblanco-us/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp: In member 

function 'void WiFiManager::handleWifiSave()':
/Users/jblanco-us/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:462:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_ip.fromString(server->arg("ip"));
                    ^
/Users/jblanco-us/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:467:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_gw.fromString(server->arg("gw"));
                    ^
/Users/jblanco-us/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:472:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_sn.fromString(server->arg("sn"));
                    ^
exit status 1
Error compiling.

domonetic, 2016-01-19:
Uipethernet is for arduino core. Wifimanager is for esp8266 core. Not is a error

yomasa, 2016-01-19:
yes, here the code

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
    
}

tzapu, 2016-01-19:
You may want to check that board type in arduino ide is set to an esp variant.
Since you are using the latest github manager you will need to checkout the latest esp8266 core for arduino as well

yomasa, 2016-01-19:
I reinstall everything fresh when I compile the AutoConnect sample I get this

Build options changed, rebuilding all
/Users/jblanco-us/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp: In member function 'void WiFiManager::handleWifiSave()':
/Users/jblanco-us/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:462:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_ip.fromString(server->arg("ip"));
                    ^
/Users/jblanco-us/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:467:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_gw.fromString(server->arg("gw"));
                    ^
/Users/jblanco-us/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:472:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_sn.fromString(server->arg("sn"));
                    ^
exit status 1
Error compiling.

tzapu, 2016-01-19:
hi, since you got the lates from github, you need to get the latest esp8266 arduino core as well

cheers
alex

yomasa, 2016-01-19:
How do i do that?

yomasa, 2016-01-19:
using 2.0

tzapu, 2016-01-19:
https://github.com/esp8266/Arduino#using-git-version-

yomasa, 2016-01-19:
Did that still same error, How can I check the versions?

mocheffendi, 2016-01-19:
What version of IDE did you use?

yomasa, 2016-01-19:
this one
http://arduino.esp8266.com/staging/package_esp8266com_index.json

yomasa, 2016-01-19:
I tried  the stable and staging versions

mocheffendi, 2016-01-19:
arduino IDE v 1.6.7 can't compile ESP 8266 sketch
use Arduino IDE v 1.6.5 to compile ESP 8266 sketch.

yomasa, 2016-01-19:
I got it working, I use Arduino 1.6.7 and had to install wifi manager 0.5.0 then 0.6.0 for it to work.

if you just install 0.6.0 I get

/Users/jblanco-us/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:467:20: error: 'class IPAddress' has no member named 'fromString'
_sta_static_gw.fromString(server->arg("gw"));
^
/Users/jblanco-us/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:472:20: error: 'class IPAddress' has no member named 'fromString'
_sta_static_sn.fromString(server->arg("sn"));
^
exit status 1
Error compiling.

mocheffendi, 2016-01-19:
after use the latest library and the latest example AutoConnectWithFSParametersAndCustomIP
i got the same error

tzapu, 2016-01-19:
hi, you need to update the esp8266 cora package, not the arduino ide itself
https://github.com/esp8266/Arduino#using-git-version-

nibelungen, 2016-01-20:
# Same here:
Arduino 1.6.7 (current)
ESP8266 2.0.0 (stable)
WiFiManger (master dl from 19.01.2016)
# Used Example AutoConnect
      same Error as yomasa
# Used Example AutoConnect (modified setAPConfig - uncommented string)

     AutoConnect:20: error: 'class WiFiManager' has no member named 'setAPConfig'

     wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));
                 ^
     exit status 1
     'class WiFiManager' has no member named 'setAPConfig'

tzapu, 2016-01-20:
setAPConfig was renamed to wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));
see examples and readme
cheers

tzapu, 2016-01-20:
but, on top of that, you will get yomasa s error, you need a newer version of the esp8266 package, the github one for now

nibelungen, 2016-01-20:
thanks for as always fast feedback. 
Maybe you want to update your examples to the new syntax - even if it is uncommented.

Will try the git version after i had tried this:

Arduino 1.6.5r5
ESP8266 2.1.0-rc1 (staging does not work with Arduino 1.6.7)
WiFiManger (master.zip dl from 19.01.2016)

# Used Example AutoConnect (unchanged)
    arduino-1.6.5-r5\portable\sketchbook\libraries\WiFiManager-master\WiFiManager.cpp:462:20:        error: 'class IPAddress' has no member named 'fromString'
         _sta_static_ip.fromString(server->arg("ip"));

tzapu, 2016-01-20:
forgot about random tests in samples, will cleanup soon i guess

you still need a newer esp8266, the fromString function call above has been added a few days ago

nibelungen, 2016-01-20:
Latest Test with

Arduino 1.6.5r5
ESP8266 current git (WiFi files from 19.01.2016 16:27)
WiFiManger (master.zip dl from 19.01.2016)

# Result
	ESP8266WiFi\ESP8266WiFiGeneric.cpp.o: In function `std::vector<WiFiEventCbList_t, std::allocator<WiFiEventCbList_t> >::size() const':
	\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.cpp:336: multiple definition of `wifi_dns_found_callback(char const*, ip_addr*, void*)'
	ESP8266WiFi\ESP8266WiFi.cpp.o:\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:729: first defined here
	ESP8266WiFi\ESP8266WiFiSTA.cpp.o: In function `ESP8266WiFiSTAClass::smartConfigDone()':
	\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp:559: multiple definition of `wifi_wps_status_cb(wps_cb_status)'
	ESP8266WiFi\ESP8266WiFi.cpp.o:\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:775: first defined here
	collect2.exe: error: ld returned 1 exit status

nibelungen, 2016-01-20:
@tzapu what is your environment (Arduino Version ...)

tzapu, 2016-01-20:
arduin ide 1.6.6
checkout of https://github.com/esp8266/Arduino#using-git-version- latest 
and obviously the latest WiFiManager :)

tzapu, 2016-01-20:
@nibelungen the error above looks like there is something wrong in your libraris/hardware folder

are you sure you don t have it included twice?

nibelungen, 2016-01-20:
Can confirm it works with

Arduino 1.6.6
ESP8266 current git
WiFiManager latest 

Thanks tzapu

mocheffendi, 2016-01-20:
Hi Tzapu,

Arduino IDE 1.6.5 r5
ESP 8266 git version
WiFiManager latest
confirm compile and upload but this is the serial monitor for sketch from the example AutoConnectWithFSParametersAndCustomIP.ino

```
mounting FS...
mounted file system
reading config file
opened config file
{"blynk_token":"bf21f65cf73546fcb90a26de7f7179d7"}
parsed json

Exception (28):
epc1=0x4000bf0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff05c0 end: 3fff09b0 offset: 01a0

>>>stack>>>
3fff0760:  3fffdad0 3fff0774 3fff09fc 40206b83  
3fff0770:  3ffe8618 3fff0770 3fff077c 00000000  
3fff0780:  3fff1c28 00000002 3fff1c34 feefeffe  
3fff0790:  feefeffe feefeffe feefeffe feefeffe  
3fff07a0:  feefeffe feefeffe feefeffe feefeffe  
3fff07b0:  feefeffe feefeffe feefeffe feefeffe  
3fff07c0:  feefeffe feefeffe feefeffe feefeffe  
3fff07d0:  feefeffe feefeffe feefeffe feefeffe  
3fff07e0:  feefeffe feefeffe feefeffe feefeffe  
3fff07f0:  feefeffe feefeffe feefeffe feefeffe  
3fff0800:  feefeffe feefeffe feefeffe feefeffe  
3fff0810:  feefeffe feefeffe feefeffe feefeffe  
3fff0820:  feefeffe feefeffe feefeffe feefeffe  
3fff0830:  feefeffe feefeffe feefeffe 00000018  
3fff0840:  3ffe9170 00000000 000003e8 feefeffe  
3fff0850:  3fff1bc0 3fff1c08 feefeffe feefeffe  
3fff0860:  feefeffe feefeffe feefeffe feefeffe  
3fff0870:  feefeffe feefeffe feefeffe feefeffe  
3fff0880:  feefeffe feefeffe feefeffe feefeffe  
3fff0890:  feefeffe feefeffe feefeffe feefeffe  
3fff08a0:  feefeffe feefeffe feefeffe feefeffe  
3fff08b0:  feefeffe feefeffe feefeffe feefeffe  
3fff08c0:  feefeffe feefeffe feefeffe feefeffe  
3fff08d0:  feefeffe feefeffe feefeffe feefeffe  
3fff08e0:  feefeffe feefeffe feefeffe feefeffe  
3fff08f0:  feefeffe feefeffe feefeffe feefeffe  
3fff0900:  feefeffe feefeffe feefeffe feefeffe  
3fff0910:  feefeffe feefeffe feefeffe feefeffe  
3fff0920:  feefeffe feefeffe feefeffe feefeffe  
3fff0930:  feefeffe feefeffe feefeffe feefeffe  
3fff0940:  feefeffe feefeffe feefeffe feefeffe  
3fff0950:  feefeffe feefeffe feefeffe feefeffe  
3fff0960:  feefeffe feefeffe feefeffe feefeffe  
3fff0970:  feefeffe feefeffe feefeffe feefeffe  
3fff0980:  feefeffe feefeffe feefeffe 3ffef988  
3fff0990:  3fffdad0 00000000 3ffef980 4020d144  
3fff09a0:  feefeffe feefeffe 3ffef990 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

tzapu, 2016-01-20:
hi,

try a SPIFFS.format() before ?

Seems to be some memory issue i guess

Did you get the latest example as well, I think i updated that one yesterday…
Cheers

mocheffendi, 2016-01-20:
yes confirm solved
thanks

mocheffendi, 2016-01-20:
hi tzapu,

no network found 

[image]

tzapu, 2016-01-21:
do you have a quality filter set? maybe it s too aggressive ..

mocheffendi, 2016-01-21:
hi Tzapu,

disable the quality filter 
and back to 0.6 with no resut on scan

```
  //set minimum quality of signal so it ignores AP's under that quality
  //defaults to 8%
  //wifiManager.setMinimumSignalQuality();
```

tzapu, 2016-01-21:
it s weird, i am not sure what else you could check or try
i ve tested it on mine as well, i am not seeing one APs that I know is there, but I am definetly seeing a lot of others

tzapu, 2016-01-21:
might be this issue here https://github.com/esp8266/Arduino/issues/1355

yomasa, 2016-01-21:
The issue is in the wifiManager code, I had to setup another computer today to work on my project, New IDE NEW install of everything, Same problems then I went into the update manager and install version 0.5.0 of wifiManager and then updated to 0.6.0 that fixed it no more errors.

mocheffendi, 2016-01-22:
use esp8266 latest github and WiFiManager latest github no result on scan wifi

back to
use esp8266 2.00 and WiFiManager 0.6 scan with result (confirm OK)

i think the bug is on esp8266 board latest github. git clone https://github.com/esp8266/Arduino.git esp8266

i'm waiting for stable release of esp8266 2.1 and WiFiManager 0.7 

thanks

tzapu, 2016-01-22:
if it s the error above, it got introduced with the update to sdk 1.5 which is not in 2.0.0, only above
i hope 2.1.0, or at least another candidate with some fixes comes out fast

bidyutper, 2016-01-30:
what version should i use...arduino ide,,,esp core & wifi manager got confusd....currently usin arduino ide 1.6.7 core 2.0 & wifi manager 6.0..plz

tzapu, 2016-01-31:
for core 2.0.0 you can only use 0.6
for core 2.1.0rc2 you can use 0.7

hope this helps

bidyutper, 2016-01-31:
wrkin with ide 1.6.5....core 2.0.0 & wifi manager 0.5.0
nxt will try with ide 1.6.6 & ide 1.6.7

tzapu, 2016-01-31:
you might try 0.6 on 2.0.0 with 1.6.5 as well

bidyutper, 2016-01-31:
yeah tht one also wrkin fine...thx for awsm config on esp ... currently usin mqtt config let me try out ths....

scropion86, 2016-01-31:
@tzapu i think you should keep the current stable version 0.7 and wait till ESP/Arduino core updated as the Newly release SDK 1.5.2 has WPS enabled so this also can be implemented in your WIFi Manager.
and Make a clear Note in your Github Readme that stable release is compatible with ESP/Arduino core 2.0.
and trunk version compatible with ESP/Arduino core RC 2.0.1 or whatever RC version

tzapu, 2016-01-31:
unfortunatelly, 0.7 does not work with 2.0.0, only with the latest staging.

after the next stable, things should be a lot better, we shall see

got any docs pertaining to WPS on the esp?

cheers

scropion86, 2016-01-31:
here it's 
http://bbs.espressif.com/viewtopic.php?f=46&t=1702
http://bbs.espressif.com/viewtopic.php?f=46&t=1703

there is also changes in the AT commands and you can create your own AT command.

tzapu, 2016-02-04:
@mocheffendi igrr just added a fix for the scanning missing access points 
it works for me
can you try as well?

flloto, 2016-02-07:
Hi, 
with ide 1.6.5....core 2.0.0 & wifi manager 0.6.0
Not working: 
Error- Similar Yomasa:
compile the AutoConnect sample:

C:\Users\Fede\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp: In member function 'void WiFiManager::handleWifiSave()':
C:\Users\Fede\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:462:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_ip.fromString(server->arg("ip"));
                    ^
C:\Users\Fede\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:467:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_gw.fromString(server->arg("gw"));
                    ^
C:\Users\Fede\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp:472:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_sn.fromString(server->arg("sn"));
                    ^
Error compiling.

with ide 1.6.5....core 2.0.0 & wifi manager 0.9.0

Work fine, I´m testing.

tzapu, 2016-02-07:
that would be correct, i have added some code in 0.8 i think to make it compile on core 2.0.0

closed 2016-02-16 by tzapu

danielitp, 2016-05-13:
error: 'class WiFiManager' has no member named 'getConfigPortalSSID'
Serial.println(myWiFiManager->getConfigPortalSSID());

error: 'class WiFiManager' has no member named 'setMinimumSignalQuality'

wifiManager.setMinimumSignalQuality(50);
error: invalid conversion from 'void ()(WiFiManager)' to 'void (*)()' [-fpermissive]

wifiManager.setAPCallback(configModeCallback);

C:\Users\itp\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:61:13: error: initializing argument 1 of 'void WiFiManager::setAPCallback(void (*)())' [-fpermissive]

 void    setAPCallback( void (*func)(void) );

tzapu, 2016-05-13:
hi,

you need to update to the latest wifimanager

cheers

YogeshKarale, 2017-06-14:
Hello,

I am using Arduino 1.8.3
Esp8266 package "https://github.com/esp8266/Arduino/releases/download/2.3.0/esp8266-2.3.0.zip"
wifimanager library  0.12.0

error: 

AutoConnect:20: error: 'class WiFiManager' has no member named 'setAPConfig'

     wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

                 ^

exit status 1
'class WiFiManager' has no member named 'setAPConfig'

YogeshKarale, 2017-06-14:
UPDATE:

It works only when I use wifimanager library 0.6.0 (V6).

Branen101, 2017-09-27:
have an issue when i try to load onto card
Arduino: 1.8.4 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

sketch\APScan.cpp: In member function 'bool APScan::start()':

APScan.cpp:18: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

   results = WiFi.scanNetworks(true, settings.apScanHidden); // (async = true & show_hidden = true)

                                                          ^

sketch\APScan.cpp:18:58: note: candidate is:

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)

     int8_t scanNetworks(bool async = false);

            ^

C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tablatronix, 2017-09-27:
What code is that? Its not wifimanager.

`results = WiFi.scanNetworks(true, settings.apScanHidden); // (async = true & show_hidden = true)`

rajrawoor, 2018-08-31:
Hi ..
I am getting the following errors when trying to compile .. Please help ...

Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp: In member function 'void WiFiManager::setupConfigPortal()':
/Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:146:71: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(), WiFiManager* const>::type)'
   server->on(String(F("/")), std::bind(&WiFiManager::handleRoot, this));
                                                                       ^
/Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:146:71: note: candidates are:
In file included from /Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:17:0,
                 from /Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:13:
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, THandlerFunction handler);
        ^
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note:   no known conversion for argument 1 from 'String' to 'const char*'
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn);
        ^
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note:   candidate expects 3 arguments, 2 provided
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn);
        ^
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 2 provided
/Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:147:81: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(unsigned char), WiFiManager* const, bool>::type)'
   server->on(String(F("/wifi")), std::bind(&WiFiManager::handleWifi, this, true));
                                                                                 ^
/Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:147:81: note: candidates are:
In file included from /Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:17:0,
                 from /Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:13:
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, THandlerFunction handler);
        ^
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note:   no known conversion for argument 1 from 'String' to 'const char*'
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn);
        ^
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note:   candidate expects 3 arguments, 2 provided
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn);
        ^
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 2 provided
/Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:148:83: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(unsigned char), WiFiManager* const, bool>::type)'
   server->on(String(F("/0wifi")), std::bind(&WiFiManager::handleWifi, this, false));
                                                                                   ^
/Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:148:83: note: candidates are:
In file included from /Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:17:0,
                 from /Users/rawoor/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:13:
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)
[... cut ...]
/Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 2 provided
Multiple libraries were found for "WiFiManager.h"
 Used: /Users/rawoor/Documents/Arduino/libraries/WiFiManager
 Not used: /Users/rawoor/Documents/Arduino/libraries/WiFiManager-master
Multiple libraries were found for "ESP8266WebServer.h"
 Used: /Users/rawoor/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer
 Not used: /Users/rawoor/Documents/Arduino/libraries/ESPWebServer-master
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

tablatronix, 2018-08-31:
Use esp 2.4+

tablatronix, 2018-10-02:
This is no longer an issue in dev version.
Will try to update hotfixes

au190, 2018-10-31:
Hi All
Is there any update on this ? I got the same error "rajrawoor commented on Aug 31" WiFiManager.cpp error.

How can I compile the code ?
Has anybody did it with success ?

Thx a lot of your answer.

tablatronix, 2018-11-01:
I will try to fix today

au190, 2018-11-02:
Oh that would be GREAT !
Can you replay here is its done and working ?
Thx