tzapu/WiFiManager #78: Any Way to add Custom Fields to be saved

yomasa, 2016-01-19:
I want to save the "mqtt_server"  , "mqtt_port", "mqtt_user","mqtt_password"  ?

Thanks
Jorge

Jorgen-VikingGod, 2016-01-20:
there is already support for custom fields:
https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino

tzapu, 2016-01-20:
thanks Jorgen , mentioned in docs here https://github.com/tzapu/WiFiManager#custom-parameters

closed 2016-01-20 by tzapu

Jorgen-VikingGod, 2016-01-20:
@tzapu Wow, did not see the updated doc yet.
Awesome work - really like what you did

tzapu, 2016-01-20:
thanks, could not have done it without all your guy's help

drvpn, 2022-06-21:
I would love to see an example where a custom field input type was password.  I have figured out how to add the field to the webpage but I cannot figure out how to store the password in a variable.  custom_field.getValue() doesn't work.

Below is  snippet  of code how I added the field.
```
const char* custom_mpass_str = "<br/><label for='pwd'>MQTT Password:</label><input type='password' id='pwd' name='pwd'>";
new (&custom_field) WiFiManagerParameter(custom_mpass_str); // custom html input

wm.addParameter(&custom_field);

```
Does anyone know how I can get the pwd's value and store it in variable?

Thanks!

tablatronix, 2022-06-21:
See the fs examples, spiffs etc

drvpn, 2022-06-21:
Thanks @tablatronix but unfortunately that example doesn't help.  Below is all the parameters that is configured in https://github.com/tzapu/WiFiManager/blob/master/examples/Parameters/SPIFFS/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino and this method only works to create input type="text" not input type="password".  I am looking to input an actual password, so people cannot shoulder surf and see the password.

```
WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 6);
WiFiManagerParameter custom_api_token("apikey", "API token", api_token, 32);
```

Since I cannot use the above method to create a field with input type="password" I have found that I can add a password field like this:

```
const char* custom_mpass_str = "<br/><label for='pwd'>MQTT Password:</label><input type='password' id='pwd' name='pwd'>";
new (&custom_field) WiFiManagerParameter(custom_mpass_str); // custom html input

```

This works but I cannot figure out how to get the value out of the webpage and into a variable.  This follow method won't work:

`strcpy(mqtt_server, custom_mpass_str.getValue());`

So looks like I am stuck here unless I can find a way to retrieve the input value, when the form is submitted, and store it in a variable.  Any ideas?

tablatronix, 2022-06-21:
you can just add `type='password'` to the custom html parameter?

drvpn, 2022-06-21:
Thanks again @tablatronix I found a solution in an example that used radio buttons.  It used the follow code and it seems to work with my custom password field as well.  Here is the function that was required:

```
String getParam(String name) {
  //read parameter from server, for customhmtl input
   String value;
  if (wm.server->hasArg(name)) {
    value = wm.server->arg(name);
  }
  return value;
}

```

Here is the link to the solution I found: https://github.com/tzapu/WiFiManager/issues/1158#issuecomment-1047708779

Thanks so much for all your time!

tablatronix, 2022-06-21:
```C++
  WiFiManagerParameter custom_input_type("input_pwd", "input pass", "", 15,"type='password'"); // custom input attrs (ip mask)
```

drvpn, 2022-06-21:
@tablatronix That works like a charm and is far more intuitive than the getParam solution. Today you are my code hero, thanks so much! 🥇