tzapu/WiFiManager #79: LED Indcator Debug

mocheffendi, 2016-01-22:
Dear Tzapu,

add feature LED indicator debug please
1. On AP mode        : Blink LED 5 times on a sec
2. On Station Mode  : Blink LED with dimmer 2 times on a sec

Thanks
Best Regards

MEf

tzapu, 2016-01-22:
hi, i thought about this, and i think i can sort it in an example, with what we have currently, no modifications to WiFiManager. Would that be ok for you?

mocheffendi, 2016-01-22:
yes, 
Thank you

domonetic, 2016-01-22:
Use the callback function for this. No need modifications.

tzapu, 2016-01-23:
as @domonetic said, you can use the call back for start config portal

added example here https://github.com/tzapu/WiFiManager/tree/master/examples/AutoConnectWithFeedbackLED

mocheffendi, 2016-01-23:
@Tzapu 
confirm work 
thank you very much

tzapu, 2016-01-23:
cool
have fun

closed 2016-01-23 by tzapu