tzapu/WiFiManager #80: Gives up too quickly

buzztiaan, 2016-01-23:
Did i read right that it only tries connecting once, and then gives up and becomes an access point?

I think some configurable retry-count is in order, with some smallish delay between then. Just to make it a bit more userfriendly on not perfect reception locations :D

tzapu, 2016-01-24:
hi, only tries once, 
but in my experience, when reception is poor, the connecting functions just stand there waiting a lot longer until they can establish the connection.
if you have those kinds of issues, maybe it s worth thinking about starting it on a button press rather than automatically... and on normal startup retry as you see fit

buzztiaan, 2016-01-24:
whats the exact issue you are avoiding by not doing something as simple as 'try 5 times' ?

tzapu, 2016-01-25:
needless delays, i am not a fan of delays :)
i ve got some battery powered sensors for instance, that could do without waiting for a number of retries 
(it could be set to 1, of course).
there are also some situations(routers i guess) in which i ve had the connection take a looong while before connecting, having 5 retries would mean it would take forever to get to see the configuration access point

buzztiaan, 2016-01-25:
i see it give up within 3 seconds as-is, in situations where all wifi clients around it have a fine signal

i'll just fork and implement, see if it gets better
the default could be repeat once ;)

tzapu, 2016-01-25:
for example, i have a router that at times takes 20 secs or more to connect or fail... i m saying 20 because i kind of give up counting by then, could be longer 30-40

tzapu, 2016-01-31:
i ve also just thought of this, using the OnDemandConfigPortal + a WiFi.begin() in a loop of retries you can achieve exactly what you propose, so the tools are already there

tzapu, 2016-02-16:
@buzztiaan i have added a new option to WiFiManger to set a connect timeout. 
in order to use this you d need to update to the latest github version of the lib and call 
`wifiManager.setConnectTimeout(30); `
before autoConnect. parameter is timeout in seconds . 
i am very intereste to know if this would help in your case

buzztiaan, 2016-02-16:
I'll test it out within 24hrs!

tzapu, 2016-02-23:
also, if that doesn t work, could you try the following sketch and see if you get a connection?
https://gist.github.com/tzapu/5b13866aaa584b32074a

tablatronix, 2016-02-29:
It looks like the core autoconnect tries 3 times, but very quickly also.

tzapu, 2016-03-01:
well, the thing is it gets the idle status set very fast, so it doesn t attempt anything else...

tzapu, 2016-03-08:
hi, i ve just made another commit to this, could you please try the github version of WiFiManager and see if anything changed for you?

buzztiaan, 2016-08-06:
been a while since i tried this

just grabbed latest version through libmanager, and all previous issues seem to not be present :D 
closing for now, will open new issue when it happens ;)

closed 2016-08-06 by buzztiaan