tzapu/WiFiManager #81: HTML UI is not shown

probonopd, 2016-01-25:
Trying to use the AutoConnectWithTimeout sketch with Arduino 1.6.8.201601190826 and esp8266/Arduino from trunk on Linux. AutoConnectAP shows up, I can connect to it, and in the browser I get redirected to 182.168.4.1 but then there is only a blank HTML page with nothing visible on it. Tested with multiple iOS devices.

Serial monitor shows:

```
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
```

probonopd, 2016-01-25:
The AutoConnect sketch even crashes the ESP when I try to reload the white page rapidly from the iPhone:

```
*WM: 
*WM: AutoConnect
*WM: Reading SSID
*WM: SSID: 
*WM: 
*WM: Reading Password
*WM: Password: 
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: 
*WM: Configuring access point... 
*WM: ESPxxx
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root

Exception (9):
epc1=0x401055b5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4a029e65 depc=0x00000000

ctx: cont 
sp: 3ffeff50 end: 3fff0400 offset: 01a0

>>>stack>>>
3fff00f0:  3fff1de0 40106fd7 3fff01d8 40106fc0  
3fff0100:  3fff1dc8 00000000 3ffe8b6b 40106fc0  
3fff0110:  3fff0190 00000000 00000010 40106fc0  
3fff0120:  3fff019c 00000000 00000010 40208b42  
3fff0130:  3ffe8b6b 00000000 3fff0190 40208b9a  
3fff0140:  3ffe8b6b 00000000 3fff0190 40208bca  
3fff0150:  0000000d 3fff0f18 3fff0190 40208c18  
3fff0160:  3ffe8b6b 3fff0ef8 3fff019c 40208c18  
3fff0170:  3ffe8b6b 3fff0ef8 3fff0190 40208a74  
3fff0180:  3fff17c4 3fff0ef8 3fff0f18 40205b4f  
3fff0190:  00000000 00000000 00000000 3fff1f08  
3fff01a0:  0000000f 00000000 3fff1ef0 0000000f  
3fff01b0:  00000000 3ffef180 00000000 00000030  
3fff01c0:  3fff1ed8 0000000f 00000000 3fff1de8  
3fff01d0:  0000000f 00000000 3fff1e48 0000001f  
3fff01e0:  00000015 3fff1dd0 0000000f 00000003  
3fff01f0:  3fff1e10 0000002f 00000022 40202e70  
3fff0200:  3fff0f34 40207a90 3ffef3e0 00000000  
3fff0210:  00000001 3fff1b50 40207b74 3ffef3e0  
3fff0220:  00000000 00000000 3fff0ef8 00000000  
3fff0230:  000000d9 3fff0f18 3fff0ef8 40204a3f  
3fff0240:  3ffef180 00000000 3fff0e98 40203c25  
3fff0250:  00000000 3fff1ea8 3fff0340 40207bec  
3fff0260:  3ffef180 00000000 3fff0340 402066d0  
3fff0270:  3fff02bc 00000000 3ffef30c 40208399  
3fff0280:  3fff05e0 40106fd7 3fff02bc 40205fdc  
3fff0290:  4020243b 00000000 3fff0340 3fff0e80  
3fff02a0:  00000000 00000000 3fff0340 40206797  
3fff02b0:  00000000 00000000 00000000 00000000  
3fff02c0:  00000000 00000000 3fff0e68 0000000f  
3fff02d0:  00000000 3fff0e50 0000000f 00000000  
3fff02e0:  3fffdad0 00000000 3fff0340 3ffef3cc  
3fff02f0:  3fffdad0 00000000 3fff0340 40206808  
3fff0300:  00000000 00000000 00000000 00000000  
3fff0310:  00000000 00000000 3fff0e80 0000000f  
3fff0320:  0000000b feefeffe 3fff0340 40205f36  
3fff0330:  3fffdad0 00000000 3ffef30c 40202018  
3fff0340:  3fff0e98 3fff0ef8 00000000 0000000a  
3fff0350:  3fff0d98 0000007f 0000007c feefeffe  
3fff0360:  3fff0e80 00000000 3fff0e20 0000000f  
3fff0370:  00000000 3fff0e38 0000000f 00000000  
3fff0380:  00000000 000017df 3ffe8cc8 00000000  
3fff0390:  3ffe8cc8 00000000 3ffe8cc8 00000000  
3fff03a0:  00000000 ffffffff 00000000 fe010035  
3fff03b0:  00000000 00000000 feefeffe feefeffe  
3fff03c0:  feefeffe feefeffe feefeffe feefeffe  
3fff03d0:  feefeffe feefeffe feefeffe feefeffe  
3fff03e0:  feefeffe feefeffe 3ffef3c5 40207bb8  
3fff03f0:  feefeffe feefeffe 3ffef3e0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

tzapu, 2016-01-26:
hi, is this with the latest trunk of WiFiManager from yesterday?

mtnbrit, 2016-01-26:
I'm getting the same result with 2.1.0-rc2. works properly with rc1

using v0.6.0 of wifimanager

tzapu, 2016-01-26:
hi, please try and get the latest wifimanager from github(as opposed to a release) if you use rc2, i made some more changes
cheers

probonopd, 2016-01-26:
Originally I had used the version from Library Manager. 

Now I have tested the version from git master, with esp8266/Arduino 2.1.0-rc2, and it is working. So to make things easier, it would be great if the new version could appear in the Library Manager. Thanks for this useful piece of software!

tzapu, 2016-01-26:
thank you for using it :)

unfortunately i can't release a version to library manager until esp8266 core 2.1 is release to stable, or people using both stable packages from lib manager, won t be able to use it...

f3z0, 2016-01-26:
Not complaining but it would be useful to bump version any time master changes, I made the mistake to assume the library was up-to-date by comparing the version numbers but I didn't actually have the latest.

tzapu, 2016-01-27:
hi, for the reason above i can t do that, people are installing it through board manager, together with esp8266 core, so those two need to work together

after esp8266 core 2.1.0 is made stable, i can do that until i start using some other function that they ve just added in the code (a few functions the great guys there added at my request for wifimanager, they are that cool)

f3z0, 2016-01-27:
I had installed core via board manager and then also WiFiManger via board
manager and it didn't work (compile errors mentioned by others) that's how
I ended up here.

probonopd, 2016-01-28:
Be aware that users can select a version from the drop-down menu in Board Manager. So imho adding a version check to the latest version that advises users of pre-2.1.0 that this version of the library can only be used with 2.1.0 or later would be sufficient for adding the latest version of the library to the Board Manager.

tzapu, 2016-01-29:
uuuh.... did not see this, 
actually i now remember, i m rather stupid, i did see this :)
does it display the decription of each version as well?
thanks for this, i will start adding it as soon as i get some time
nice

probonopd, 2016-01-29:
It shows the version number. A description is what you could put it to appear at compile time...

tzapu, 2016-01-30:
but if the library does not even compile, what currently happens with 0.7 and 2.0.0 ... can it show the description?
i tried, the description in boards manager is not updated

probonopd, 2016-01-30:
Well I think you could use compile-time `#ifdef` and/or ``if statements to print some error if the dependencies are not correct. Just an idea, haven't verified so far.

probonopd, 2016-01-31:
I asked @igrr if there is an easy way to detect the esp8266/Arduino version and give a meaningful error message if it is too low. This is his response:
> there isn't, but that's a good idea to add. unfortunately if we add it now (i.e. `ARDUINO_ESP8266_2_1_0`), you won't be able to distinguish between older versions (i.e. 2.0.0)
there are some tricks which you can use though - for instance, you may find some member function (in any of the core classes) which was added around the same time as the function you need and then use `SFINAE` to trigger code depending on the presence of given member function see e.g. this for example: http://stackoverflow.com/a/87846/5659071 and then use `static_assert` to show meaningful error message

tzapu, 2016-01-31:
uu, that s brilliant, thanks and i ll try and see if i can get my head around it :D

tzapu, 2016-02-01:
thanks @probonopd 
i went down the rabbit hole of c++ and i ve updated WiFiManger to 0.8. which should let everyone compile down to esp8266 core 2.0.0
it will also throw a warning in the serial console, if the function called is not supported by the current core
cool :P

tzapu, 2016-02-04:
i will now close this issue, feel free to reopen it if you feel it s needed

closed 2016-02-04 by tzapu