tzapu/WiFiManager #82: password issue

ratzl, 2016-01-26:
When I try to enter a password in the configuration page, it's not working when the
password contains the + sign

example password:  +Test-    (not working)
example password:  -Test-    ( working)

shirish47, 2016-01-27:
well I am not sure whats the problem with me when I try to connect after configuration it fails even though password is correct... it seems that there is some extra character.. or something .. my password contains alphabets and a number and "("  ")"..... previous libraries used to work...

tzapu, 2016-01-27:
thanks for the reports, 
@ratzl & @shirish47 could you please let me know how your password ends up in the url string after save? it should be in the p= var

shirish47, 2016-01-27:
yes it comes that way ..accept "(" turn to %28 and ")" to %29  I think that is normal even when wifi trys to connect it shows that password correctly in the debug msgs.

shirish47, 2016-01-27:
*WM: Sent config page
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
*WM: Request redirected to captive portal
*WM: Handle root

shirish47, 2016-01-27:
when I reset it I get this ..
*WM: AutoConnect
*WM: Reading SSID
*WM: SSID: 
*WM: UNICORN
*WM: Reading Password
*WM: Password: Un1c()rn
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 1
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

tzapu, 2016-01-27:
are you using 0.6 release ? or latest from github
what esp8266 core version are you using?

shirish47, 2016-01-27:
version 2 for esp8266 core and for this library its 0.6

ratzl, 2016-01-28:
*WM: 
*WM: AutoConnect
*WM: Reading SSID
*WM: SSID: 
*WM: rapi-wlan
*WM: Reading Password
*WM: Password: -12Li5a3-       //password was -12Li5a3-
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 134.103.68.138

*WM: 
*WM: AutoConnect
*WM: Reading SSID
*WM: SSID: 
*WM: rapi-wlan
*WM: Reading Password
*WM: Password:  12Li5a3-      //password was +12Li5a3-
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 4
*WM: 
*WM: Configuring access point... 
*WM: ESP125778
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

domonetic, 2016-01-28:
What value returns getPassword()?
there may be a bug in the password storage from portal

domonetic, 2016-01-28:
here the error:
in urldecore if char == '+'

if (*src == '+') {
      decoded += ' ';
      *src++;
    } else {

shirish47, 2016-01-29:
ok can you explain the error? is it encoding it before saving to file and then decoding it and having trouble. but my password does not include any other characters other than "(" and ")"...

shirish47, 2016-01-29:
this library's old (0.5) version works fine with my code.

tzapu, 2016-01-29:
guys, sorry about this, i m away until middle of next week and can only look/do at minor stuff, will try looking more next week

shirish47, 2016-01-29:
no problem buddy .. thanks :)

domonetic, 2016-01-29:
@shirish47
I really do not understand that urldecode , but I think that's the problem with special characters.

shirish47, 2016-01-29:
ya its I think something related to Base 64 I am not sure .. but I had similar issue with base 64 ..one of servers were not accepting my data after encoded to b64, because of null termination.. and its difficult to handle strings in C without null termination.

tzapu, 2016-02-04:
i think i removed the + sign at some point for a specific purpose that i now can t remember :(
i think it may have been messing with spaces in ssids ... really can t remember
i think there might be a new version of url decoding coming soon as part of the core... i ll look into adding that

tzapu, 2016-02-04:
hi all, this should be sorted in 0.9
i was double decoding as the server lib was already doing that
please test and reopen if needed

closed 2016-02-04 by tzapu