tzapu/WiFiManager #83: Suggestion:  List networks in strength priority

MrBill1234, 2016-01-28:
When selecting 'configure wifi' I think it may be good to list the SSID's in a more organised fashion.

List % signal strength first, then Locked or unlocked, then the SSID name.  Then order by Signal strength.  Reasoning being that the strongest one is probably close to you and one of yours.    It would also neaten up the screen a bit :)  Not sure if an HTML table is needed - shouldn't be.

Just my 2p.

tzapu, 2016-02-16:
hi, long on the table to do this, but most stuff that s easily done just increases the file size and memory usage of the library... not ideal in an embedded system

tzapu, 2016-02-16:
there s part of it (the aspect part) modified in the last commit

tzapu, 2016-02-16:
sorted networks as well in 0.10
enjoy and test :D

closed 2016-02-16 by tzapu

MrBill1234, 2016-02-19:
Looks good - one thing I found, if one of the SSID's is too long, and you're on a small screen (iPhone 5c),  it seems to break the table a bit for subsequent SSID's listed.  Might be good to only display first 25 to 27 characters of the SSID to make it look neater.  In particular some of the HP wifi enabled printers seem to have some pretty long SSID's.

tzapu, 2016-02-19:
hi, got any screenshots of this?
thought it should wrap nicely, but apparently not..

MrBill1234, 2016-02-19:
[image]

tzapu, 2016-02-19:
cheers, i see what you mean