tzapu/WiFiManager #84: parameter update

yomasa, 2016-01-28:
I need to update stored parameter in the code after connecting, is there a function to do this.?

If I need to write my own how to I get the address range  used by wifimanager to I do not over write anything?

tzapu, 2016-01-29:
hi, which sketch are you using? just ssid and pass or other stuff as well?

yomasa, 2016-01-29:
ssid and pass only, also I am still get this; just installed on my laptop new everything, can you tell me or point me on what it is I need to update. I using 2.0 ver of esp ide or is there something else?

Arduino: 1.6.7 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

/Users/jorge/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp: In member function 'void WiFiManager::handleWifiSave()':
/Users/jorge/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:472:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_ip.fromString(server->arg("ip"));
                    ^
/Users/jorge/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:477:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_gw.fromString(server->arg("gw"));
                    ^
/Users/jorge/Documents/Arduino/libraries/WiFiManager-master/WiFiManager.cpp:482:20: error: 'class IPAddress' has no member named 'fromString'
     _sta_static_sn.fromString(server->arg("sn"));
                    ^
exit status 1
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

tzapu, 2016-01-30:
with version 2.0.0 of esp8266 core use version 0.6 of wifimanager from board manager

cheers

tzapu, 2016-02-01:
i ve updated WiFiManger to 0.8. might sort some of your issues, try it if you can

tzapu, 2016-02-04:
if you use just ssid and pass, there is no config.json. they are saved internally by the sdk

scropion86, 2016-02-16:
is that means the other parameters like MQTT or Blynk Token will be lost if ESP restarted?

tzapu, 2016-02-16:
if they are saved to fs like in example, no

closed 2018-01-26 by tablatronix